pimatic/rfcontroljs #28: Request : (new) protocol for Intertechno Shutterswitch CMR500

RenGro, 2015-01-12:
Hi,

I have a Intertechno shutter CMR-500 as well as remote control ITZ-500.
The shutter is working in pilight by using "elro-hc" protocol. But pilight is only able to open and close the shutter completely - not stopping the shutter in the middle of the window...

So I tried to control the shutter by pimatic and available class": "HomeduinoRFShutter" . I tried all switching protocols but without success :-(

This problem is equal to an earlier issue from "leader21". But the solution for switch4 protocol is not working in my case. 

That' s why I used the 'RF receive' command to track the RF signal from remote control.
Systemcode A - unitcode 1 --> State on
0101 0101 0101 0101 0101 0101 0101 0101 0101 0110 0110 0110 02
Systemcode A - unitcode 2 --> State on
0101 0101 0101 0101 0110 0101 0101 0101 0101 0110 0110 0110 02
Systemcode A - unitcode 3 --> State on
0101 0101 0101 0101 0101 0110 0101 0101 0101 0110 0110 0110 02
Systemcode A - unitcode4 --> State on
0101 0101 0101 0101 0110 0110 0101 0101 0101 0110 0110 0110 02
Systemcode A - unitcode 16 --> State on
0101 0101 0101 0101 0110 0110 0110 0110 0101 0110 0110 0110 02
Systemcode B - unitcode 1 --> State on
0110 0101 0101 0101 0101 0101 0101 0101 0101 0110 0110 0110 02
Systemcode B - unitcode 2 --> State off
0110 0101 0101 0101 0110 0101 0101 0101 0101 0110 0110 0101 02

The schema seems to be as follow
pulseCount: 50
systemcode: 0-3
unicode:  4-7
?: 8 --> fix 0101 
?: 9 --> fix 0110
?:10 --> fix 0110
state: 11

When we use '0' as "0101" and '1' / 'F' as '0110', so we can use the following overview for system- and unit codes
http://www.fhemwiki.de/wiki/Intertechno_Code_Berechnung

According to this schema I'm able to generate each sequence for systemcode and unitcode manually and control the shutter by the 'RF send' on command line, e.g. 
RF send 4 3 376 1074 9964 0 0 0 0 0 1100101010101010110010101010101010101100110010102

Could you please support me for choosing the right (available) protocol or could you please implement a new suitable protocol?

Thanks in advance!

leader21, 2015-01-13:
Hi RenGro!
My intertechno shutter devices are working very well with pimatic-homeduino.
I don't think that Intertechno has several RF protocols for their devices.
Have you set your turnwheel codes like at the table shown here ? The pimatic codes differ from the pilight ones.
https://github.com/pimatic/pimatic/issues/380

my cmr500 device for the Code A 11 looks like that : 
```
    {
      "id": "beamer-leinwand",
      "name": "Beamer Leinwand",
      "class": "HomeduinoRFShutter",
      "protocols": [
        {
          "name": "switch4",
          "options": {
            "unit": 31,
            "id": 10
          }
        }
      ]
    },
```
Never had any issues.

RenGro, 2015-01-13:
Hi leader21,

thanks for your quick answer.
But it doesn't work :-(

A11 configured as described by you
 {
      "id": "homeduino-contact4",
      "name": "Shutter Controller4",
      "class": "HomeduinoRFShutter",
      "protocols": [
        {
          "name": "switch4",
          "options": {
            "unit ": 31,
	    "id": 10
          }
        }
      ]
    },
RF receive by pimatic, state: down
0110 0110 0110 0110 0110 0110 0101 0110 0101 0110 0110 0101 02

RF receive by remote control, state: off--> shutter is working
0101 0101 0101 0101 0101 0110 0101 0110 0101 0110 0110 0101 02

In this example the first 5 pulse sequences are inverted

Any ideas?

Thanks in advance!
René

closed 2015-01-13 by RenGro