pimatic/rfcontroljs #29: DRU Heater support

sweebee, 2015-01-13:
Hi,

I have a DRU Heater at home. It has a 433Mhz remote that homeduino can receive. I tried to make a protocol with rfcontroljs, but i can't get it to work. Is there someone who can help or create a protocol?

I receive codes like this: 

RF receive 298 644 26356 0 0 0 0 0 01101110011010101001101010101010011012"
debug: received: [ 298, 644, 26356 ] 01101110011010101001101010101010011012

and this is the remote (4 buttons).
[image]

Icesory, 2015-01-13:
Post pls the output for all buttons

sweebee, 2015-01-13:
Well, i receive different things every time:

set button does not send rf. The other buttons i have to hold to receive rf.

Up:
00:35:28.674 [pimatic-homeduino] debug: data: "RF receive 330 629 26379 0 0 0 0 0 01100110011010101001101010101010011012"
00:35:28.718 [pimatic-homeduino] debug: received: [ 330, 629, 26379 ] 01100110011010101001101010101010011012
00:35:28.808 [pimatic-homeduino] debug: data: "RF receive 311 647 26385 0 0 0 0 0 01100110011010101001101010101010011012"
00:35:28.852 [pimatic-homeduino] debug: received: [ 311, 647, 26385 ] 01100110011010101001101010101010011012
00:35:29.029 [pimatic-homeduino] debug: data: "RF receive 319 648 218 26412 0 0 0 0 01100110011010101001121010101010011013"
00:35:29.071 [pimatic-homeduino] debug: received: [ 218, 319, 648, 26412 ] 12211221122121212112202121212121122123

Down:
00:36:40.432 [pimatic-homeduino] debug: data: "RF receive 324 636 26337 0 0 0 0 0 01100110011010101001101010010101100112"
00:36:40.472 [pimatic-homeduino] debug: received: [ 324, 636, 26337 ] 01100110011010101001101010010101100112
00:36:40.565 [pimatic-homeduino] debug: data: "RF receive 315 643 26369 0 0 0 0 0 01100110011010101001101010010101100112"
00:36:40.606 [pimatic-homeduino] debug: received: [ 315, 643, 26369 ] 01100110011010101001101010010101100112
00:36:40.963 [pimatic-homeduino] debug: data: "RF receive 322 639 26367 0 0 0 0 0 01100110011010101001101010010101100112"
00:36:41.005 [pimatic-homeduino] debug: received: [ 322, 639, 26367 ] 01100110011010101001101010010101100112

Off: 
00:37:22.303 [pimatic-homeduino] debug: data: "RF receive 325 637 26358 0 0 0 0 0 01100110011010101001101010101001101012"
00:37:22.343 [pimatic-homeduino] debug: received: [ 325, 637, 26358 ] 01100110011010101001101010101001101012
00:37:22.439 [pimatic-homeduino] debug: data: "RF receive 310 648 26359 0 0 0 0 0 01100110011010101001101010101001101012"
00:37:22.490 [pimatic-homeduino] debug: received: [ 310, 648, 26359 ] 01100110011010101001101010101001101012
00:37:22.570 [pimatic-homeduino] debug: data: "RF receive 303 659 26391 0 0 0 0 0 01100110011010101001101010101001101012"
00:37:22.613 [pimatic-homeduino] debug: received: [ 303, 659, 26391 ] 01100110011010101001101010101001101012

koffienl, 2015-01-14:
Do you have a temperatur on your display? If so, chance is that it is not sending up or down command but actual temperature.

sweebee, 2015-01-14:
Yes there is. The up and down buttons steer the gas valve. So maybe a different code for every step?
But the temperature shows the actual room temperature. it doesn't change when pushing a button.

I tried to put the valve down to the max, but then the rf code still changes.

Icesory, 2015-01-14:
I don't see a rolling code. Homeduino is more sensitive as it needed. Pimatic filters the wrong timings.

sweebee, 2015-01-15:
I see, so the code is always the same but the timings are wrong? Is there a way to add this to homeduino? This DRU remote?

Icesory, 2015-01-16:
I think I can do this at the weekend. Can you change some addresses on the devices? It would make it easier

sweebee, 2015-01-16:
This are all the addresses, i tried to assign new codes, but its always the same. But a friend has the same remote, so reading out that remote wil help too?

Here are my addresses:

On (Off and Up together)
07:49:17.966 [pimatic-homeduino] debug: received: [ 321, 639, 26346 ] 01100110011010101001101010101001011012
07:49:18.063 [pimatic-homeduino] debug: data: "RF receive 306 654 26376 0 0 0 0 0 01100110011010101001101010101001011012"

Off
07:47:11.475 [pimatic-homeduino] debug: data: "RF receive 319 643 26327 0 0 0 0 0 01100110011010101001101010101001101012"
07:47:11.516 [pimatic-homeduino] debug: received: [ 319, 643, 26327 ] 01100110011010101001101010101001101012

Up
07:48:26.426 [pimatic-homeduino] debug: data: "RF receive 323 639 26356 0 0 0 0 0 01100110011010101001101010101010011012"
07:48:26.465 [pimatic-homeduino] debug: received: [ 323, 639, 26356 ] 01100110011010101001101010101010011012

Down
07:49:04.332 [pimatic-homeduino] debug: data: "RF receive 325 637 26323 0 0 0 0 0 01100110011010101001101010010101100112"
07:49:04.373 [pimatic-homeduino] debug: received: [ 325, 637, 26323 ] 01100110011010101001101010010101100112

sweebee, 2015-01-16:
Additional info i followed:

A code is selected automatically for all Mertik Maxitrol electronics from among 65,000 random codes available. The receiver has to learn the code of the handset:
 • Press and hold the receiver’s resetbutton until you hear two (2) acoustic signals. After the sec- ond, longer acoustic signal, release the reset button.

 • Within the subsequent 20 seconds press the (small flame) button on the remote handset until you hear an additional long signal confirming the code is set. 

NOTE: This is a one time setting only, and is not required when changing the batteries in the remote or receiver.

Icesory, 2015-01-16:
Yes pls post the codes from the other remote. Only with a different I can see the right adress coding

sweebee, 2015-01-16:
Unfortunately the remote was different, and not 433Mhz. I will try to find a different one.

sweebee, 2015-01-17:
Okay, i got the same remote from a neighbor, this are the codes from that one:

On
16:47:06.326 [pimatic-homeduino] debug: data: "RF receive 305 615 23020 0 0 0 0 0 01101001011010100110011001101001011012"
16:47:06.363 [pimatic-homeduino] debug: received: [ 305, 615, 23020 ] 01101001011010100110011001101001011012

Off
16:47:34.123 [pimatic-homeduino] debug: data: "RF receive 292 619 22983 0 0 0 0 0 01101001011010100110011001101001101012"
16:47:34.173 [pimatic-homeduino] debug: received: [ 292, 619, 22983 ] 01101001011010100110011001101001101012

Up
16:47:44.835 [pimatic-homeduino] debug: data: "RF receive 303 611 22968 0 0 0 0 0 01101001011010100110011001101010011012"
16:47:44.873 [pimatic-homeduino] debug: received: [ 303, 611, 22968 ] 01101001011010100110011001101010011012

Down
16:47:58.180 [pimatic-homeduino] debug: data: "RF receive 284 627 23076 0 0 0 0 0 01101001011010100110011001101010100112"
16:47:58.219 [pimatic-homeduino] debug: received: [ 284, 627, 23076 ] 01101001011010100110011001101010100112

Icesory, 2015-01-17:
The informatios are a little bit less. So can make a switch protocol for that should work. But when later some other guy need this protocol too and we got more informations the protocol and the configuration can be changed.

Icesory, 2015-01-17:
I have start the first version for the DRU protocol. You can download it from my git repository or you wait until @sweetpi has merged it into the official release.
https://github.com/Icesory/rfcontroljs/commit/64566ea19253f00a1adbac5f46392952e71641a3#diff-1

As device you should use the HomeduinoRFButton Device.
https://github.com/pimatic/pimatic-homeduino#buttons-device-example

sweebee, 2015-01-17:
Thank you! I tried it but it didn't work?

Up received:
18:02:57.487 [pimatic-homeduino] debug: data: "RF receive 327 635 26351 0 0 0 0 0 01100110011010101001101010101010011012"
18:02:57.527 [pimatic-homeduino] debug: received: [ 327, 635, 26351 ] 01100110011010101001101010101010011012
18:02:57.565 [pimatic-homeduino] debug: switch9:  { id: 168, unit: 64, state: true }

in config:

{
	      "id": "up",
	      "text": "+",
	      "protocols": [{
	        "name": "switch9",
	        "options": {
	          "unit": 64,
	          "id": 168,
	          "state": true
	        }
	      }]
	    }

but when i press it, nothing happens.

Icesory, 2015-01-17:
can you post your complete buttons device? And use pls markdown. https://help.github.com/articles/github-flavored-markdown/#fenced-code-blocks
```
```json 
{
    "your": "config"
}
+```  with out the +

```

looks like this
```json 
{
    "your": "config"
}
```

sweebee, 2015-01-17:
```
{
      "id": "kachel-keuken",
      "name": "Kachel keuken",
      "class": "HomeduinoRFButtonsDevice",
      "buttons": [
	      {
          "id": "uit",
          "text": "Uit",
          "protocols": [
            {
              "name": "switch9",
              "options": {
                "unit": 65,
                "id": 168,
                "state": false
              }
            }
          ]
        },
        {
          "id": "aan",
          "text": "Aan",
          "protocols": [
            {
              "name": "switch9",
              "options": {
                "unit": 65,
                "id": 168,
                "state": true
              }
            }
          ]
        },
        {
          "id": "up",
          "text": "▲",
          "protocols": [
            {
              "name": "switch9",
              "options": {
	            "id": 168,
                "unit": 64,
                "state": true
              }
            }
          ]
        },
        {
          "id": "down",
          "text": "▼",
          "protocols": [
            {
              "name": "switch9",
              "options": {
                "unit": 71,
                "id": 168,
                "state": false
              }
            }
          ]
        }
      ]
    },
```

Icesory, 2015-01-17:
The config looks right. How do installed the new protocol? And get you a rf message in pimatic, when you press a button

sweebee, 2015-01-17:
I get 'Done' in pimatic when i push a button. And [pimatic-homeduino] debug: data: "ACK"
 in the log file.

I downloaded al your files from your frcontroljs and uploaded it to:

pimatic-app/node_modules/pimatic-homeduino/node_modules/homeduino/node_modules/rfcontroljs/

[image]

Icesory, 2015-01-17:
Have you the possibility to check the rf message with a second arduino?

sweebee, 2015-01-17:
unfortunately not.

Icesory, 2015-01-18:
you can make a fast test. go into the folder
pimatic-app/node_modules/pimatic-homeduino/node_modules/homeduino/node_modules/rfcontroljs/lib
and open the file switch9.js
go to this line
pulseLengths: [300, 600, 23000]
and replace it with this
pulseLengths: [320, 640, 26330]

sweebee, 2015-01-18:
Replaced it, but still not working.

Icesory, 2015-01-18:
So late online xD. 
Pimatic-homeduino works so far? Do you use it with other RF stuff? (I try to locate the failure)

sweebee, 2015-01-18:
Yea, just back from night work.
I use it with 4 Kaku switches, 3 kaku dimmers, 1 kaku pir and one doorbel. all working great.

Icesory, 2015-01-18:
Nice.
Can you flash the the raw example from RFControl on the arduino. You must set the receiver pin in the code an upload it to the arduino. After this open a terminal program and look on the serial port for communication (pair:8N1 baud:11520). When you press now a button you get a lot of numbers this I need. Best for all buttons. On, off, up, down

sweebee, 2015-01-18:
Im not using an arduino, I use a Attiny45.

Icesory, 2015-01-18:
The Attiny is an active filter? It is highly recommended to use a Arduino. It reduces the cpu load by 40%

sweebee, 2015-01-18:
yes as an active filter, my cpu load is about 5%

Icesory, 2015-01-18:
Ah I don't own a active filter. But it gives the possibility to get the raw code. @sweetpi how can he do it?

sweebee, 2015-01-18:
Looks like i have to make a Raspberry Pi with transmitter/receiver for a friend this week, so when I have finished that, i could read the rf code that pimatic is sending for the heater.

sweetpi, 2015-01-18:
You could modify the virtualhomeduino code to print the raw timings: https://github.com/pimatic/virtualhomeduino/blob/master/vhduino.cpp#L82-L89

This is the c program that is used if you don't use a arduino. It's like the serial connection but it just reads from the stdinput. Ofcause it only works on the raspberry pi.

sweebee, 2015-01-20:
i got this from plight-raw (raw data), don't know if you can use it.

```
433gpio: 197
433gpio: 723
433gpio: 190
433gpio: 724
433gpio: 333
433gpio: 205
433gpio: 667
433gpio: 630
433gpio: 414
433gpio: 541
433gpio: 26304
433gpio: 317
433gpio: 648
433gpio: 650
433gpio: 380
433gpio: 317
433gpio: 630
433gpio: 733
433gpio: 292
433gpio: 187
433gpio: 686
433gpio: 749
433gpio: 282
433gpio: 756
433gpio: 136
433gpio: 725
433gpio: 320
433gpio: 532
433gpio: 292
433gpio: 474
433gpio: 513
433gpio: 644
433gpio: 297
433gpio: 675
433gpio: 287
433gpio: 664
433gpio: 295
433gpio: 676
433gpio: 289
433gpio: 711
433gpio: 242
433gpio: 765
433gpio: 319
433gpio: 203
433gpio: 714
433gpio: 656
433gpio: 350
433gpio: 548
433gpio: 26310
433gpio: 296
433gpio: 651
433gpio: 647
433gpio: 456
433gpio: 255
433gpio: 664
433gpio: 568
433gpio: 343
433gpio: 407
433gpio: 492
433gpio: 731
433gpio: 221
433gpio: 685
433gpio: 282
433gpio: 683
433gpio: 274
433gpio: 674
433gpio: 296
433gpio: 331
433gpio: 706
433gpio: 631
433gpio: 375
433gpio: 722
433gpio: 136
433gpio: 728
433gpio: 250
433gpio: 630
433gpio: 369
433gpio: 679
433gpio: 222
433gpio: 658
433gpio: 314
433gpio: 374
433gpio: 622
433gpio: 647
433gpio: 252
433gpio: 679
433gpio: 26310
433gpio: 293
433gpio: 662
433gpio: 646
433gpio: 292
433gpio: 358
433gpio: 834
433gpio: 502
433gpio: 335
433gpio: 247
433gpio: 691
433gpio: 595
433gpio: 285
433gpio: 673
433gpio: 313
433gpio: 649
433gpio: 291
433gpio: 698
433gpio: 266
433gpio: 428
433gpio: 556
433gpio: 646
433gpio: 393
433gpio: 635
433gpio: 268
433gpio: 713
433gpio: 312
433gpio: 652
433gpio: 277
433gpio: 617
433gpio: 519
433gpio: 504
433gpio: 301
433gpio: 237
433gpio: 765
433gpio: 617
433gpio: 528
433gpio: 334
433gpio: 26330
433gpio: 295
433gpio: 663
433gpio: 647
433gpio: 290
433gpio: 346
433gpio: 731
433gpio: 623
433gpio: 422
433gpio: 154
433gpio: 640
433gpio: 649
433gpio: 357
433gpio: 651
433gpio: 279
433gpio: 640
433gpio: 324
433gpio: 719
433gpio: 220
433gpio: 337
433gpio: 666
433gpio: 690
433gpio: 214
433gpio: 798
433gpio: 274
433gpio: 624
433gpio: 240
433gpio: 670
433gpio: 407
433gpio: 609
433gpio: 449
433gpio: 479
433gpio: 311
433gpio: 530
433gpio: 700
433gpio: 356
433gpio: 344
433gpio: 630
433gpio: 26334
433gpio: 304
433gpio: 655
433gpio: 652
433gpio: 378
433gpio: 354
433gpio: 579
433gpio: 605
433gpio: 434
433gpio: 199
433gpio: 656
433gpio: 647
433gpio: 282
433gpio: 674
433gpio: 287
433gpio: 681
433gpio: 415
433gpio: 552
433gpio: 286
433gpio: 340
433gpio: 706
433gpio: 652
433gpio: 217
433gpio: 683
433gpio: 291
433gpio: 704
433gpio: 332
433gpio: 602
433gpio: 308
433gpio: 678
433gpio: 321
433gpio: 603
433gpio: 294
433gpio: 365
433gpio: 646
433gpio: 603
433gpio: 339
433gpio: 798
433gpio: 26157
433gpio: 292
433gpio: 660
433gpio: 635
433gpio: 435
433gpio: 302
```

sweebee, 2015-02-18:
I have a arduino now and i receive this from the remote:

Up:
```
13:00:58.232 [pimatic-homeduino] debug: data: "RF receive 326 638 22472 0 0 0 0 0 0110100101100110011010101001101010101010011012"
13:00:58.233 [pimatic-homeduino] debug: received: [ 326, 638, 22472 ] 0110100101100110011010101001101010101010011012
13:00:58.365 [pimatic-homeduino] debug: data: "RF receive 305 659 22412 0 0 0 0 0 0110100101100110011010101001101010101010011012"
13:00:58.366 [pimatic-homeduino] debug: received: [ 305, 659, 22412 ] 0110100101100110011010101001101010101010011012
```
Down:
```

13:01:34.074 [pimatic-homeduino] debug: data: "RF receive 324 639 22460 0 0 0 0 0 0110100101100110011010101001101010010101100112"
13:01:34.074 [pimatic-homeduino] debug: received: [ 324, 639, 22460 ] 0110100101100110011010101001101010010101100112
13:01:34.207 [pimatic-homeduino] debug: data: "RF receive 331 645 250 22424 0 0 0 0 0112120101120110011010101001101010010101100113"
13:01:34.207 [pimatic-homeduino] debug: received: [ 250, 331, 645, 22424 ] 1220201212201221122121212112212121121212211223
```
Off:
```
13:02:06.940 [pimatic-homeduino] debug: data: "RF receive 326 638 22460 0 0 0 0 0 0110100101100110011010101001101010101001101012"
13:02:06.941 [pimatic-homeduino] debug: received: [ 326, 638, 22460 ] 0110100101100110011010101001101010101001101012
13:02:07.073 [pimatic-homeduino] debug: data: "RF receive 308 657 22460 0 0 0 0 0 0110100101100110011010101001101010101001101012"
13:02:07.075 [pimatic-homeduino] debug: received: [ 308, 657, 22460 ] 0110100101100110011010101001101010101001101012
```
On:
```
13:02:23.214 [pimatic-homeduino] debug: data: "RF receive 323 641 22468 0 0 0 0 0 0110100101100110011010101001101010101001011012"
13:02:23.214 [pimatic-homeduino] debug: received: [ 323, 641, 22468 ] 0110100101100110011010101001101010101001011012
13:02:23.347 [pimatic-homeduino] debug: data: "RF receive 327 661 250 22436 0 0 0 0 0012120101100110011010101001101010101001011013"
13:02:23.347 [pimatic-homeduino] debug: received: [ 250, 327, 661, 22436 ] 1120201212211221122121212112212121212112122123
```

Icesory, 2015-02-19:
Now i lost a little bit the overview, Can you please describe exactly what you received with which system and from which remote?

sweebee, 2015-02-19:
I received that codes directly from the remote with Pimatic and an arduino. I used the attiny before. I'm nog using the attiny anymore.

Icesory, 2015-02-19:
The new ones are 8 unit longer then the old ones.
```
01100110011010101001101010101001011012 old
01101001 01100110011010101001101010101001011012 new
```

sweebee, 2015-03-05:
This could be the issue because my attiny didn't work very well?

Icesory, 2015-03-05:
Are you able to make a serial connection to the arduino?

When yes. 
Stop pimatic.
Start a SerialMonitor
Send this command
```
RF send 4 PIN 323 641 22468 0 0 0 0 0 0110100101100110011010101001101010101001011012
```
replace PIN with the pin number from your sender on the arduino.

sweebee, 2015-03-05:
I opened the monitor. and received READY, but i can't type??

Icesory, 2015-03-05:
With a linux monitor i cant help you. for this you musst google.
On windows you can use the arduino IDE or HTerm

sweebee, 2015-03-05:
I tried this: echo -e "RF send 4 PIN 323 641 22468 0 0 0 0 0 0110100101100110011010101001101010101001011012" > /dev/ttyUSB0

and serial sends ACK but heater does nothing

Icesory, 2015-03-05:
PIN! You must change this.

sweebee, 2015-03-05:
echo -e "RF send 4 4 323 641 22468 0 0 0 0 0 0110100101100110011010101001101010101001011012" > /dev/ttyUSB0

output: ACK

but no response from heater.

Icesory, 2015-03-05:
Try this, it is the on command
```
RF send 4 4 323 641 22468 0 0 0 0 0 01100110011010101001101010101001011012
```

sweebee, 2015-03-05:
Nothing happens. when I try a code from a light (kaku switch) it works with the light. but the code from the heater does nothing to the heater.

Icesory, 2015-03-23:
Can you please capture the raw timings with pilight new? and please make the files bigger. I need more data to verify the code.

sweebee, 2015-03-23:
Pilight-raw you mean? Is that possible with homeduino?

Icesory, 2015-03-23:
the data from pilight-raw is very good. i can use it.

sweebee, 2015-03-23:
I have installed pilight but how do i get pilight-raw communicate with homeduino?

Icesory, 2015-03-23:
oh. no this isnt working xD.
I have a homduino version with raw output,
https://github.com/Icesory/RFControl
https://github.com/Icesory/homeduino

Best is to put the arduino on an windows pc and start the serial monitor. Set the baudrate to 115200.
Send this command.
"RF raw PIN"

sweebee, 2015-03-23:
Is there a easy way to install it through the arduino IDE? I copied all the libraries to the library folder but it keeps saying it can't find it? 

```
In file included from homeduino.ino:7:0:
rfcontrol_command.h:1:21: fatal error: RFControl: No such file or directory
 #include <RFControl>
```

sweebee, 2015-03-23:
Installed it via the raspberry. connected it to my mac and getting READY in the serial monitor. But when i send ```RF raw 0``` i get nothing.

Icesory, 2015-03-23:
next to the baudrate dropdown is a other dropdown.
Chose there the option with "NL" and "CR"
When you send the command "PING" you should got PING as answer

sweebee, 2015-03-23:
Yes, i get a PING

sweebee, 2015-03-23:
Got it working now, tried a different receiver

sweebee, 2015-03-23:
UP: http://pastebin.com/2itgMGKu
Down: http://pastebin.com/qJwapTh6
Off: http://pastebin.com/U41T70pn
On: http://pastebin.com/g3L41YSA

Icesory, 2015-03-23:
That is very good. In the next days i will look into this data. I hope that i can get it working.

sweebee, 2015-03-23:
On the raw reading; it worked with my receiver that was soldered on a breadboard. I tried 2 different receivers on a pin breadboard but that didn't work? Tried the receivers with a different 433mhz protocol and they are working fine.

Icesory, 2015-03-23:
hmm that is interesting. It is possible that a poor receiver receives 868MHz data as 433MHz. It is a harmonic of this. 
Now you can try a little bit by your self.
Connect the arduino to the mac and use the Serial monitor to send some data.
Syntax is the following:
"RF send 'PIN' 'REPEATS' '8BUCKETS' 'DATA'"
Pin is the number of the digital pin from the transmitter. 
Repeats should be something between 2 and 7. Try there some different.
8Buckets are allways this "326 638 22460 0 0 0 0 0"
And the Data are the rf message to send.
Feel free to try some messages. When the heater response to one than in think we get it working.

[27 more replies]

closed 2015-03-24 by Icesory