pimatic/rfcontroljs #32: Elro Home Easy HE877 switch problem

Takisg, 2015-01-17:
Elro Home Easy HE877 switch working with protocol "switch 2".
I have a problem with the receiving mode.
On receiver mode decode the off command from the remote control but the on command decode the same result off.

Debug 
ON
switch3: { houseCode: 22, unitCode: 10, state: false }
switch2: { houseCode: 22, unitCode: 10, state: false }

OFF
switch3: { houseCode: 22, unitCode: 10, state: true }
switch2: { houseCode: 22, unitCode: 10, state: false }

Any help......

Icesory, 2015-01-17:
try this
https://github.com/pimatic/pimatic/issues/476#issuecomment-70384443

Takisg, 2015-01-18:
I try your solution....

Now with the modify protocol switch4 i relearn the switch and it is working.
But the state in the gui not change anymore.

The strange is that the log says true and false for the received state....

Xento, 2015-02-20:
I think this is the same problem that I had: https://github.com/pimatic/pimatic-homeduino/issues/30

closed 2015-04-01 by Icesory

RaspiGuru, 2017-05-07:
Hi Guys,
I would like to control my HomeEasy HE877A model wireless plugs with my Raspberry equiped with a  433Mhz emitter.
I am new to this, could you please anyone help me to install it? I would like to be know the requirements (libraries and plugins) and installation procedures. 
With regards

leader21, 2017-05-07:
You need the homeduino plugin. Check out howto section to.learn more. If you need any further assistance please use our forum.pimatic.org thx!
https://forum.pimatic.org/topic/202/4-homeduino-433-mhz-sending-receiving-and-even-more