pimatic/rfcontroljs #33: Alecto WS-3500

sweebee, 2015-01-22:
Can anyone confirm that this sensor works with pimatic?

Alecto W-044

http://www.weerstationkopen.nl/accessoires-weerstations/alecto/w-044-sensor

Icesory, 2015-01-22:
Confirm not but i think we can i implement this when it isn´t yet.

sweebee, 2015-01-22:
I need 2 temperature meters for a thermostat, so i have to buy 2. Do you know one that will work definitely? I have a Cresta TX320 lying around but pimatic won't receive anything at all (Its 433 mhz).

Icesory, 2015-01-22:
Its a hint from @leader21 
http://www.ebay.de/itm/Logilink-WS0002-Outdoor-Sensor-einheit-fur-Wireless-Wetterstation-WS0001-/131098170352

sweebee, 2015-01-22:
Under which weather device should i add that one to the config?

leader21, 2015-01-22:
the logilink is added via weather1 protocol.
here's my config : 
```
    {
      "id": "wettersensor1",
      "name": "Temperatur Balkon",
      "class": "HomeduinoRFTemperature",
      "protocols": [
        {
          "name": "weather1",
          "options": {
            "id": 141
          }
        }
      ]
    },
``` 
the ID will be shown at the homeduino debug output. 
after you determined the ID of the weather device you can add it to the config and finally to the GUI's page where you want it to.

sweebee, 2015-01-22:
Thank you! I ordered 2.

leader21, 2015-01-22:
feel free to share your experiences with us :)

sweebee, 2015-01-22:
Looking at the weather1 protocol, it says battery status. Is this supported by this sensor? Because i use them to measure the temperature and control the heating with a dummy thermostat. So when the battery is almost empty it can send me a notification, otherwise it would be a problem to control the temperature in the room.

leader21, 2015-01-22:
the sensor is transmitting this information but it's not finally implemented. maybe icesory is able to change the protocol or whatever is necessary.
the information is received but not shown at the gui as far as i know. you cannot use this information at the moment. 
really cool would be an option like <code>IF weathersensor.battery is $bad THEN push title:"Battery low" message:"Battery of weathersensor is low" priority:1
so you will be noticed via pushover message for example or whatever you think of :)

closed 2015-03-23 by Icesory