pimatic/rfcontroljs #35: Add FA20RF support

Xento, 2015-02-08:
Could you add support for FA20RF smoke detecors?
It is already supported by fhemduino, there you could get the protocol from.
I looked at your code but I don't really understand where to implement the different timings.
Is there any documentation?

Icesory, 2015-02-09:
Ca you post the protocol Dokumentation? Think next weekend I got time to make some work

Xento, 2015-02-09:
Here is the file used by fhemduino.
https://github.com/mdorenkamp/fhemduino/blob/master/lib/FA20RF.cpp

Xento, 2015-02-15:
I think I understood how it works.
I will try it by myself.
But we need a special device definiton in pilight because it acts like a pir when receiving.
It sends the alarm every 2-3 seconds when on and this stops when the alarm turns off.
When you want to send alarm you have to send the alarmsignal every 2-3 seconds so that the smokedector stays on.

leader21, 2015-02-15:
what about protocol name "alarm1" ?
i'm having this device too, so i could check if your protocol works for me also.

sweetpi, 2015-02-15:
> But we need a special device definiton in pilight because it acts like a pir when receiving.

ups pilight?

In pimatic we can either abuse a pir-sensor (there is an autoreset option) or we can create a new alarm device.

Xento, 2015-02-15:
The best would be a new alarm device where you could set an interval for repeating the signal.

Xento, 2015-02-15:
When I press the test button on one smokedetector I receive ths from homeduino:
received: [ 744, 1464, 2807, 8092, 17292 ] 3002020101010102020201010101020101010102020202010204

Look at my fork for the changes I did.
https://github.com/Xento/rfcontroljs/blob/master/lib/controller.js
https://github.com/Xento/rfcontroljs/blob/master/lib/protocols/alarm1.js

In my eyes it should work now, but it doesn't.

Icesory, 2015-02-15:
Why do you want to encode a message? Think this isn't necessary.

Icesory, 2015-02-15:
And you killed a zero. 
....010-204 
You have 204 declared as a nothing but I think the it is one bit more to decode. So only 04 is the header

Xento, 2015-02-15:
The decoding is nessesary to detect which smokedetector sends the alarm.
Yeah you were right, I missed a bit.
Now receiving works but sending not.
This is received when i press the testbutton: 3002020101010102020201010101020101010102020202010204
And this is send when sending via pimatic: 0020201010101020202010101010201010101020202020102003
The 3 at the beginning and 03 at the end should only be 4.

Icesory, 2015-02-15:
The decoding is necessary. But why do you want to send from pimatic? How should this work? 

But it looks right.

Xento, 2015-02-15:
You could use the smokedetecors as alarmsirens.
For example if you are not at home and one windowsensor detects that a window was opened.

Icesory, 2015-02-15:
That is a nice feature.
I have look in to the fhemduino protocol and your implementation is right. Strange why that you get a wrong encoded message.

Xento, 2015-02-17:
Finaly I got it working: https://github.com/pimatic/rfcontroljs/pull/40
I can understand coffescript a bit, normaly I try to copy&paste things and use try&error to get the coffescript working.
Could you create the coffescript?

Xento, 2015-02-17:
I made a rule, to continuosly send an alarm:
if SomeSwitch is turned on or [SomeSwitch is on and smokedetecor is turned off] then switch smokedetector  on and after 4 seconds switch smokedetector off

Xento, 2015-02-19:
@leader21 could you test the protocol, too?

Icesory, 2015-02-21:
I will implent this to pimatic but i think the coffeescript ist not right. Why do you need a loop in the encoding part?

Xento, 2015-02-21:
The rf signal must be repeated a few times until the smokedetecros responses.
I thought a loop would be much nicer and more dynamic than repeating the pulse string.
In fhemduino the repetion is configureable.

Icesory, 2015-02-21:
Homeduino sends the RF message normally 7 times. I have merged your work into my reporsitory. can you take a look on this? I made some changes to fit it into the pimatic coding style

Xento, 2015-02-21:
Without the loop I have to switch the device on and off for about 3 or 4 times until it goes on.

Xento, 2015-02-21:
And your version sends the signal when turning off the switch.

Xento, 2015-02-21:
Double post?
I used your changes.

Icesory, 2015-02-21:
Oh yes double post. Sorry
I have found the failure while sending

Xento, 2015-02-21:
Ok, this looks better, but the problem with the sending/receiving is still there.
I have to turn the switch on and of multiple times until it works.

Icesory, 2015-02-21:
And now?

Xento, 2015-02-21:
Now it works

Icesory, 2015-02-21:
Nice. And the other protocols?

Xento, 2015-02-21:
Home Easy Advanced works, too.
I can't test the belldoor, because it was a user request and I don't have one

closed 2015-03-23 by Icesory