pimatic/rfcontroljs #36: pir recognized as switch

xCite1986, 2015-02-09:
Hello.

My Name is Alex and i use a raspberry Pi B+ with 433mhz sender & reciever with pimatic and homeduino-plugin.

I'm now getting pilight-less with your plugin and my old+new intertechnos and weathersensors works fine. many thanks for your awesome work ;)

ill see your homeduino-plugin has 2 pir-protocolls (pir1, pir2) included and i would use one of them with this PIR-sensor:

http://www.importe-uzman.de//funkalarmanlagen-bewegungsmelder-kompatibel-zonen-anlage-p-4476.html?osCsid=0cd6b4f19f537ee2fd0a0afcf6a33f30

my problem was the rfcontroljs recognized the device as two-protocoll-switch, but i need them as pir

debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
debug [pimatic-homeduino]: received: [ 170, 514, 5318 ] 01010110011001010110011001100101101010100101010102
debug [pimatic-homeduino]: data: "RF receive 170 514 5318 0 0 0 0 0 01010110011001010110011001100101101010100101010102"

I try to add following lines in my config - but useless and give me error outputs

{
  "id": "homeduino-pir",
  "name": "PIR",
  "class": "HomeduinoRFPir",
  "protocols": [
    {
      "name": "switch6",
      "options": {
        "systemcode": 9,
        "programcode": 28
      }
    }
  ],
  "resetTime": 6000
}
Debuglog:
error [pimatic]: Error loading device homeduino-pir: "switch6" is not a pir protocol.

{
  "id": "homeduino-pir2",
  "name": "PIR2",
  "class": "HomeduinoRFSwitch",
  "protocols": [
    {
      "name": "switch6",
      "options": {
        "systemcode": 9,
        "programcode": 28
      }
    }
  ],
  "resetTime": 6000
}
Makes me a Switch in pimatic and Debuglog:
error [pimatic]: Error loading device homeduino-pir: "switch6" is not a pir protocol.

Maybe you could help me with a new plugin for this type PIR-sensor?

Many thanks in advance and sorry for my bad english.
-Greedings from Austria - 
Alex

Icesory, 2015-02-09:
This type of pir sensor protocol isn't implemented at the moment. Eventually has @Yves911 time to implement this. He is the pir expert.

Yves911, 2015-02-09:
@xCite1986 hi, maybe the first thing to do is to check if you have some jumpers in your PIR sensor.
Could you take some pictures of the inside of the PIR and share it with us here?
Could you also try to find some technical documentation about it?

xCite1986, 2015-02-09:
hey @Yves911! thanks for your quick reply. 

i have many jumpers in there: 4 pins for distance, 3 pins for pulse (count), 3 pins for "usetest" (?) and a pinset to change systemcode/programcode currently switch6/8

[image]
[image]
[image]
[image]
[image]

heres the debug output of currently setting shown from pictures:

debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:37:17debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:37:17debug [pimatic-homeduino]: received: [ 167, 517, 5321 ] 01010110011001010110011001100101101010100101010102
20:37:17debug [pimatic-homeduino]: data: "RF receive 167 517 5321 0 0 0 0 0 01010110011001010110011001100101101010100101010102"
20:37:16debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:37:16debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:37:16debug [pimatic-homeduino]: received: [ 169, 515, 5318 ] 01010110011001010110011001100101101010100101010102
20:37:16debug [pimatic-homeduino]: data: "RF receive 169 515 5318 0 0 0 0 0 01010110011001010110011001100101101010100101010102"
20:37:16debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:37:16debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:37:16debug [pimatic-homeduino]: received: [ 168, 516, 5322 ] 01010110011001010110011001100101101010100101010102
20:37:16debug [pimatic-homeduino]: data: "RF receive 168 516 5322 0 0 0 0 0 01010110011001010110011001100101101010100101010102"
20:37:16debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:37:16debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:37:16debug [pimatic-homeduino]: received: [ 168, 516, 5321 ] 01010110011001010110011001100101101010100101010102
20:37:16debug [pimatic-homeduino]: data: "RF receive 168 516 5321 0 0 0 0 0 01010110011001010110011001100101101010100101010102"

if i change PULSE-Jumper from 1 to 2:

debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:38:41debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:38:41debug [pimatic-homeduino]: received: [ 170, 515, 5323 ] 01010110011001010110011001100101101010100101010102
20:38:41debug [pimatic-homeduino]: data: "RF receive 170 515 5323 0 0 0 0 0 01010110011001010110011001100101101010100101010102"
20:38:41debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:38:41debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:38:41debug [pimatic-homeduino]: received: [ 169, 515, 5302 ] 01010110011001010110011001100101101010100101010102
20:38:41debug [pimatic-homeduino]: data: "RF receive 169 515 5302 0 0 0 0 0 01010110011001010110011001100101101010100101010102"
20:38:41debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:38:41debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:38:41debug [pimatic-homeduino]: received: [ 169, 514, 5314 ] 01010110011001010110011001100101101010100101010102
20:38:41debug [pimatic-homeduino]: data: "RF receive 169 514 5314 0 0 0 0 0 01010110011001010110011001100101101010100101010102"
20:38:41debug [pimatic-homeduino]: switch8: { systemcode: 18, programcode: 'A4', state: false }
20:38:41debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:38:41debug [pimatic-homeduino]: received: [ 171, 513, 5317 ] 01010110011001010110011001100101101010100101010102
20:38:41debug [pimatic-homeduino]: data: "RF receive 171 513 5317 0 0 0 0 0 01010110011001010110011001100101101010100101010102"

if i change the use-test jumper to the upper pins, nothing will happened. the device wont send...

the pinset is corrently set to:

1 hi
4 hi
8 hi

set it to: 

1 hi
4 lo
8 hi

the debug shows:

debug [pimatic-homeduino]: switch8: { systemcode: 16, programcode: 'A4', state: false }
20:46:37debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
20:46:37debug [pimatic-homeduino]: received: [ 169, 514, 5325 ] 01010110011010100110011001100101101010100101010102
20:46:37debug [pimatic-homeduino]: data: "RF receive 169 514 5325 0 0 0 0 0 01010110011010100110011001100101101010100101010102"
20:46:37debug [pimatic-homeduino]: switch8: { systemcode: 16, programcode: 'A4', state: false }
20:46:37debug [pimatic-homeduino]: switch6: { systemcode: 9, programcode: 28, state: true }
[... 14 lines cut ...]
4 hi
6 lo

debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
20:49:05debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
20:49:05debug [pimatic-homeduino]: received: [ 170, 514, 5323 ] 01101010011010100110010101100110101010100101010102
20:49:05debug [pimatic-homeduino]: data: "RF receive 170 514 5323 0 0 0 0 0 01101010011010100110010101100110101010100101010102"
20:49:04debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
20:49:04debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
20:49:04debug [pimatic-homeduino]: received: [ 169, 515, 5324 ] 01101010011010100110010101100110101010100101010102
20:49:04debug [pimatic-homeduino]: data: "RF receive 169 515 5324 0 0 0 0 0 01101010011010100110010101100110101010100101010102"
[... cut ...]
20:49:04debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
20:49:04debug [pimatic-homeduino]: received: [ 169, 515, 5310 ] 01101010011010100110010101100110101010100101010102
20:49:04debug [pimatic-homeduino]: data: "RF receive 169 515 5310 0 0 0 0 0 01101010011010100110010101100110101010100101010102"

hope this outputs will help

i think it will be hard to find a technical documentation of this ho-name-china-pir ;) it was a component of the alarm system in the link:

http://www.amazon.de/Funk-Alarmanlage-16-Zonen-Komplettsystem/dp/B002YK1J0O

Yves911, 2015-02-09:
Okay the could you pls make 1 additional test : remove jumper on pulse count and post result here...

xCite1986, 2015-02-09:
set to 2 hi, 4 hi, 6 lo; pulse open

debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
22:03:07debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
22:03:07debug [pimatic-homeduino]: received: [ 169, 514, 5345 ] 01101010011010100110010101100110101010100101010102
22:03:07debug [pimatic-homeduino]: data: "RF receive 169 514 5345 0 0 0 0 0 01101010011010100110010101100110101010100101010102"
22:03:07debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
22:03:07debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
22:03:07debug [pimatic-homeduino]: received: [ 170, 219, 512, 5339 ] 02202020022020200221020202200220202020200202020203
22:03:07debug [pimatic-homeduino]: data: "RF receive 170 512 219 5339 0 0 0 0 01101010011010100112010101100110101010100101010103"
22:03:07debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
22:03:07debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
22:03:07debug [pimatic-homeduino]: received: [ 152, 186, 513, 5344 ] 02212121022121210221020202210221212121210202020203
22:03:07debug [pimatic-homeduino]: data: "RF receive 152 513 186 5344 0 0 0 0 01121212011212120112010101120112121212120101010103"
22:03:06debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
22:03:06debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
22:03:06debug [pimatic-homeduino]: received: [ 169, 240, 511, 5355 ] 02202020022020200220020202200220202120200202020203
22:03:06debug [pimatic-homeduino]: data: "RF receive 169 511 240 5355 0 0 0 0 01101010011010100110010101100110101210100101010103"

Yves911, 2015-02-09:
Ok so if i read correctly, homeduino see only switch6 or switch8 whatever configuration you put your jumpers.
I think it won't be a bid deal to create the protocol for it (i will give it a try this weekend) [or you can try by yourself using those instructions https://github.com/pimatic/pimatic/issues/222#issuecomment-56294356]

PS : pls post the raw data or log using the code functionality with \``` before and after the code it will be easier to read, here is an example.
```
code
```

xCite1986, 2015-02-09:
thanks for this information :+1: 

i will give it a try tomorrow with your manual to edit the switch6.js, edit the name: 'switch6' to 'pir3', type: 'switch', to 'pir' and hope this would work. ^^

Yves911, 2015-02-09:
Yes it's a good start, with "gulp test" you'll be able to check things before you propose a patch...

xCite1986, 2015-02-09:
"gulp test" makes me a lot of errors

what i have done:

sudo git clone https://github.com/pimatic/rfcontroljs.git
cd rfcontroljs
npm install
npm install --global gulp
gulp test

```
pi@raspberrypi ~/rfcontroljs $ gulp test
[23:16:03] Using gulpfile /home/pi/rfcontroljs/gulpfile.js
[23:16:03] Starting 'test'...

  #decodePulses()
    ✓ generic should decode the pulses (47ms)
    ✓ pir1 should decode the pulses (40ms)
Warning trying to parse message with protocol switch2: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch2.coffee:27:23)
Warning trying to parse message with protocol switch3: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch3.coffee:27:23)
Warning trying to parse message with protocol switch4: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch4.coffee:27:23)
Warning trying to parse message with protocol switch7: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch7.coffee:26:23)
Warning trying to parse message with protocol pir1: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/pir1.coffee:27:23)
    1) pir2 should decode the pulses
    ✓ weather1 should decode the pulses
    ✓ weather2 should decode the pulses
    ✓ weather3 should decode the pulses
    ✓ weather4 should decode the pulses
    ✓ weather5 should decode the pulses
    ✓ dimmer1 should decode the pulses
    ✓ switch1 should decode the pulses
    ✓ switch2 should decode the pulses
    ✓ switch4 should decode the pulses
Warning trying to parse message with protocol switch2: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch2.coffee:27:23)
Warning trying to parse message with protocol switch3: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch3.coffee:27:23)
Warning trying to parse message with protocol switch4: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch4.coffee:27:23)
Warning trying to parse message with protocol switch7: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch7.coffee:26:23)
Warning trying to parse message with protocol pir1: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/pir1.coffee:27:23)
Warning trying to parse message with protocol switch2: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch2.coffee:27:23)
Warning trying to parse message with protocol switch3: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch3.coffee:27:23)
Warning trying to parse message with protocol switch4: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch4.coffee:27:23)
Warning trying to parse message with protocol switch7: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch7.coffee:26:23)
Warning trying to parse message with protocol pir1: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/pir1.coffee:27:23)
Warning trying to parse message with protocol switch2: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch2.coffee:27:23)
Warning trying to parse message with protocol switch3: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch3.coffee:27:23)
Warning trying to parse message with protocol switch4: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch4.coffee:27:23)
Warning trying to parse message with protocol switch7: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch7.coffee:26:23)
Warning trying to parse message with protocol pir1: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/pir1.coffee:27:23)
Warning trying to parse message with protocol switch2: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pi/rfcontroljs/src/protocols/switch2.coffee:27:23)
[... 124 lines cut ...]

  33 passing (10s)
  1 failing

  1) #decodePulses() pir2 should decode the pulses:
     Error: timeout of 2000ms exceeded
    at done (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runnable.js:205:67)
    at callFn (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runnable.js:261:7)
    at Test.Runnable.run (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runnable.js:243:7)
    at Runner.runTest (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runner.js:373:10)
    at /home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runner.js:451:12
    at next (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runner.js:298:14)
    at /home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runner.js:308:7
    at next (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runner.js:246:23)
    at Object._onImmediate (/home/pi/rfcontroljs/node_modules/gulp-mocha/node_modules/mocha/lib/runner.js:275:5)
    at processImmediate [as _immediateCallback] (timers.js:330:15)

[23:16:26] 'test' errored after 23 s
[23:16:26] Error in plugin 'gulp-mocha'
Message:
    1 test failed.
pi@raspberrypi ~/rfcontroljs $
```

so whats wrong?

xCite1986, 2015-02-10:
many thanks for your support, now it works without nodecompilling.

first create manually a new protocoll in 
pimatic-app/node_modules/pimatic-homeduino/node_modules/homeduino/node_modules/rfcontroljs/lib/protocols/pir3.js

```
module.exports = function(helper) {
  var binaryToPulse, protocolInfo, pulsesToBinaryMapping;
  pulsesToBinaryMapping = {
    '0101': '1',
    '1010': '1',
    '0110': '0',
    '02': ''
  };
  binaryToPulse = {
    '0': '0110',
    '1': '1010'
  };
  binaryToPulse2 = {
    '0': '0110',
    '1': '0101'
  };
  return protocolInfo = {
    name: 'pir3',
    type: 'pir',
    values: {
      systemcode: {
        type: "number"
      },
      programcode: {
        type: "number"
      },
      presence: {
        type: "boolean"
      }
    },
    brands: [],
    pulseLengths: [150, 453, 4733],
    pulseCount: 50,
    decodePulses: function(pulses) {
      var binary, result;
      binary = helper.map(pulses, pulsesToBinaryMapping);
      return result = {
        unit: helper.binaryToNumber(binary, 0, 4),
        id: helper.binaryToNumber(binary, 5, 9),
        presence: true
      };
    }
  };
};

```

then i bind it to pimatic-app/node_modules/pimatic-homeduino/node_modules/homeduino/node_modules/rfcontroljs/lib/controller.js

after restart pimatic, debug log shows me following output:

```
debug [pimatic-homeduino]: pir3: { unit: 10, id: 19, presence: true }
23:50:35debug [pimatic-homeduino]: switch8: { systemcode: 0, programcode: 'A16', state: false }
23:50:35debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 25, state: true }
23:50:35debug [pimatic-homeduino]: received: [ 170, 518, 5319 ] 01101010011010100110010101100110101010100101010102
```

maybe not the best protocoll-quality, but works for me now ;-)

PS: pir3.js works also with this type of contact-sensor:

http://www.importe-uzman.de//funk-alarmanlagen-glasbruchmelder-kompatibel-zonen-anlage-p-4479.html?osCsid=0cd6b4f19f537ee2fd0a0afcf6a33f30

[... 34 lines cut ...]
00:04:45debug [pimatic-homeduino]: switch8: { systemcode: 2, programcode: 'A0', state: true }
00:04:45debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 30, state: true }
00:04:45debug [pimatic-homeduino]: received: [ 116, 154, 454, 4806 ] 12202121122112121220122121212121212121212121212113
00:04:45debug [pimatic-homeduino]: data: "RF receive 154 454 116 4806 0 0 0 0 01121010011001010112011010101010101010101010101003"
00:04:45debug [pimatic-homeduino]: pir3: { unit: 10, id: 15, presence: true }
00:04:45debug [pimatic-homeduino]: switch8: { systemcode: 2, programcode: 'A0', state: true }
00:04:45debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 30, state: true }
00:04:44debug [pimatic-homeduino]: received: [ 149, 207, 452, 4813 ] 02202020022002020220022020202020202020202020212003
00:04:44debug [pimatic-homeduino]: data: "RF receive 149 452 207 4813 0 0 0 0 01101010011001010110011010101010101010101010121003"
00:04:44debug [pimatic-homeduino]: pir3: { unit: 10, id: 15, presence: true }
00:04:44debug [pimatic-homeduino]: switch8: { systemcode: 2, programcode: 'A0', state: true }
00:04:44debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 30, state: true }
00:04:44debug [pimatic-homeduino]: received: [ 151, 454, 4773 ] 01101010011001010110011010101010101010101010101002
00:04:44debug [pimatic-homeduino]: data: "RF receive 151 454 4773 0 0 0 0 0 01101010011001010110011010101010101010101010101002"
00:04:44debug [pimatic-homeduino]: pir3: { unit: 10, id: 15, presence: true }
00:04:44debug [pimatic-homeduino]: switch8: { systemcode: 2, programcode: 'A0', state: true }
00:04:44debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 30, state: true }
00:04:44debug [pimatic-homeduino]: received: [ 118, 153, 455, 4809 ] 12202021122112121221122121212121212121212121212113
00:04:44debug [pimatic-homeduino]: data: "RF receive 153 455 118 4809 0 0 0 0 01121210011001010110011010101010101010101010101003"
00:04:44debug [pimatic-homeduino]: pir3: { unit: 10, id: 15, presence: true }
00:04:44debug [pimatic-homeduino]: switch8: { systemcode: 2, programcode: 'A0', state: true }
00:04:44debug [pimatic-homeduino]: switch6: { systemcode: 10, programcode: 30, state: true }
00:04:44debug [pimatic-homeduino]: received: [ 124, 161, 455, 4808 ] 12202020122012121220122021212121212121212121212103
00:04:44debug [pimatic-homeduino]: data: "RF receive 161 455 124 4808 0 0 0 0 01121212011201010112011210101010101010101010101023"
```

Icesory, 2015-02-11:
Next step is to clone the git repo of the rfcontroljs and add your pir3 protocol. Normally all the coding work is done in coffeescript so you get 2 new files pir3.coffee and pir3.js and the extended controller.coffee and controller.js files. When all files are added you can open a pull request and all the code will be implemented in your name.

Icesory, 2015-03-24:
What is the actual state of your work?
@xCite1986 Do you need some help?

xCite1986, 2016-02-14:
hey ronny.

sorry for late answering... 

i take a little time for response (lol - yes i know... nearly one year - but i take time for learn coding and github'ing)

i created pir6.js and pull my changes.. 

greedings from vienna

closed 2016-02-14 by xCite1986