pimatic/rfcontroljs #42: RF RGB controller support

goudenvis, 2015-03-03:
Can anyone build a protocol for a RGB controller? I receive the raw code in pimatic (in combination with the pimatic-homeduino plugin), but i'm unable to control it.
This is the RGB controller: http://www.ebay.com/itm/12V-Touch-Screen-Dimmable-Remote-Wireless-RF-Controller-For-Led-RGB-Module-Strip/371239945160?_trksid=p3693.c100102.m2452&_trkparms=ao%3D1%26asc%3D20140212121249%26meid%3D066cfc0809fc45f7b07629d56234a045%26pid%3D100102%26

I saved for every button the code:
On:
debug [pimatic-homeduino]: contact2: { id: 1035831, contact: false }
23:38:04debug [pimatic-homeduino]: switch5: { id: 1035831, unit: 0, all: true, state: true }
23:38:04debug [pimatic-homeduino]: received: [ 344, 981, 1312, 9464 ] 01010101010110100101011010100101100101010110010203
23:38:04debug [pimatic-homeduino]: data: "RF receive 344 981 1312 9464 0 0 0 0 01010101010110100101011010100101100101010110010203"
23:38:04debug [pimatic-homeduino]: received: [ 345, 967, 9484 ] 01010101010110100101011010100101100101010110011102
23:38:04debug [pimatic-homeduino]: data: "RF receive 345 967 9484 0 0 0 0 0 01010101010110100101011010100101100101010110011102"

Off:
debug [pimatic-homeduino]: received: [ 346, 966, 9476 ] 01010101010110100101011010100101100101010110011102
23:38:31debug [pimatic-homeduino]: data: "RF receive 346 966 9476 0 0 0 0 0 01010101010110100101011010100101100101010110011102"
23:38:31debug [pimatic-homeduino]: received: [ 347, 966, 9448 ] 01010101010110100101011010100101100101010110011102
23:38:31debug [pimatic-homeduino]: data: "RF receive 347 966 9448 0 0 0 0 0 01010101010110100101011010100101100101010110011102"

Dim:
debug [pimatic-homeduino]: contact2: { id: 1035831, contact: false }
23:39:06debug [pimatic-homeduino]: switch5: { id: 1035831, unit: 0, all: true, state: true }
23:39:06debug [pimatic-homeduino]: received: [ 342, 982, 1320, 9476 ] 01010101010110100101011010100101100101010101100203
23:39:06debug [pimatic-homeduino]: data: "RF receive 342 982 1320 9476 0 0 0 0 01010101010110100101011010100101100101010101100203"

bright:
debug [pimatic-homeduino]: received: [ 346, 966, 9484 ] 01010101010110100101011010100101100101010101101102
23:39:22debug [pimatic-homeduino]: data: "RF receive 346 966 9484 0 0 0 0 0 01010101010110100101011010100101100101010101101102"
23:39:22debug [pimatic-homeduino]: received: [ 332, 524, 979, 9440 ] 12110212120220200202122020200202201202020202202103
23:39:22debug [pimatic-homeduino]: data: "RF receive 524 979 332 9440 0 0 0 0 01002101012112122121011212122121120121212121121023"

change-color (upper button):
debug [pimatic-homeduino]: contact2: { id: 1035831, contact: false }
23:41:04debug [pimatic-homeduino]: switch5: { id: 1035831, unit: 4, all: false, state: false }
23:41:04debug [pimatic-homeduino]: received: [ 341, 983, 1328, 9500 ] 01010101010110100101011010100101100101010110100203
23:41:04debug [pimatic-homeduino]: data: "RF receive 341 983 1328 9500 0 0 0 0 01010101010110100101011010100101100101010110100203"

change-color (lower button):
debug [pimatic-homeduino]: received: [ 344, 970, 9464 ] 01010101010110100101011010100101100101010110101102
23:41:59debug [pimatic-homeduino]: data: "RF receive 344 970 9464 0 0 0 0 0 01010101010110100101011010100101100101010110101102"

color wheel Red:
debug [pimatic-homeduino]: received: [ 346, 968, 9504 ] 01010101010110100101011010100101010101101010011102
23:43:44debug [pimatic-homeduino]: data: "RF receive 346 968 9504 0 0 0 0 0 01010101010110100101011010100101010101101010011102"

color wheel Green:
debug [pimatic-homeduino]: received: [ 345, 969, 9504 ] 01010101010110100101011010100101010110101001101102
23:45:02debug [pimatic-homeduino]: data: "RF receive 345 969 9504 0 0 0 0 0 01010101010110100101011010100101010110101001101102"

color wheel Blue:
debug [pimatic-homeduino]: contact2: { id: 1035839, contact: false }
23:45:59debug [pimatic-homeduino]: switch7: { unit: 3, id: 1, state: true }
23:45:59debug [pimatic-homeduino]: switch5: { id: 1035839, unit: 0, all: true, state: true }
23:45:59debug [pimatic-homeduino]: received: [ 341, 1000, 9492 ] 01010101010110100101011010100101010101010110010102
23:45:59debug [pimatic-homeduino]: data: "RF receive 341 1000 9492 0 0 0 0 0 01010101010110100101011010100101010101010110010102"

On the color wheel it was tricky, because i'm not sure if the color I picked is only reb/green/blue.

Icesory, 2015-03-03:
This looks like the new led protocol´s
Can you test this version of rfcontroljs?
https://github.com/Icesory/rfcontroljs/tree/command_protocols

goudenvis, 2015-03-09:
I have installed the newest version of the homeduino plugin (0.8.45). With the led1 protocol I can turn the controller on, but not off. So it's partial working. 

In the config I use this: 
{
      "id": "ledKeuken",
      "name": "Leds keuken",
      "class": "HomeduinoRFSwitch",
      "protocols": [
        {
          "name": "led1",
          "options": {
            "id": 2071663,
            "unit": 3
          }
        }
      ]
    }

When I turn off the controller by the remote, I see registration of the raw code, but not any protocol in the log file. But when I turn the controller off in Pimatic, I don't even see any raw code. The only registration in the log is:

debug [pimatic-homeduino]: data: "ACK"

Icesory, 2015-03-23:
I have pushed a commit for this.
You can try it by a manual download or you wait until it is in the official build.
But i am not sure that this is probably working.
There are some strange thing in the data.
https://github.com/pimatic/rfcontroljs/commit/55fcc88469e02e6b33dbfd28cc23405177546dbf

goudenvis, 2015-03-31:
Thank you so much, it works! 
Finally got some time today to try. With some trial and error, I managed to let it work. I use now the led4 protocol:
{
      "id": "ledKeuken",
      "name": "Leds keuken",
      "class": "HomeduinoRFButtonsDevice",
      "buttons": [
        {
          "id": "on-button",
          "text": "Aan",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "code:10000100",
                "id": 796
              }
            }
          ]
        },
        {
          "id": "off-button",
          "text": "Uit",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "on/off",
                "id": 796
              }
            }
          ]
        },
        {
          "id": "buttonPlus",
          "text": "+",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "bright+",
                "id": 796
              }
            }
          ]
        },
        {
          "id": "buttonMin",
          "text": "-",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "bright-",
                "id": 796
              }
            }
          ]
        },
        {
          "id": "buttonBlue",
          "text": "Blauw",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "code:00000101",
                "id": 796
              }
            }
          ]
        }
      ]
    }

I'm now trying to figure out what codes are used for the different colors. So I can make buttons for the colors I want to use.

closed 2015-04-01 by Icesory