pimatic/rfcontroljs #43: strange behavior in lib-controller.coffee

Icesory, 2015-03-06:
I develop a new protocol for the Quigg switches.
To verify this i wrote a normal test in the lib.

But it looks like the test don´t call the decodePulses function on an normal way.
For this protocol we need to erase the first pulse.
My first try was to set the first pulse to 2. A two is mapped to nothing.
```
  pulsesToBinaryMapping = {
    '10': '1' #binary 1
    '01': '0' #binary 0
    '2' : ''  #footer
  }
    decodePulses: (pulses) ->
      pulses[0] = 2
      binary = helper.map(pulses, pulsesToBinaryMapping)
```
but this fails in the test. For me is this really strange.
The first pulse isn´t set to two.

I have found this solution
```
    decodePulses: (pulses) ->
      binary = helper.map(pulses[1..], pulsesToBinaryMapping)
```
@sweetpi do i have a error in reasoning?

sweetpi, 2015-03-06:
I think you've to use: `pulses[0] = '2'` because the check 

```
if data.substr(i, search.length) is search
```

inside the map function is only true if `search` and the substring have the same types.

Icesory, 2015-03-06:
Oh damit. Clearly the two must be a Char. 
Coding at 3:00am isn't a good idea. 

But it is interesting that the Char 0 in the array was not changed and no failure was dropped.

closed 2015-03-06 by Icesory