pimatic/rfcontroljs #45: Conrad S3318P (TFA) not recognized

sysdev41, 2015-03-16:
Hi,

I tested the sensor with pilight before, it worked using the following protocol: 
http://wiki.pilight.org/doku.php/tfa_v6_0  (The photo in the pilight wiki is showing exactly my sensor model).

Here is what i get from pimatic-homeduino debug log when i press the "TX" button on the sensor:

22:58:13.427 [pimatic-homeduino] debug: data: "RF receive 598 1875 3785 7610 0 0 0 0 0101010201010201020101020101010202010201010101020201010102010101020102010201010101010303"
22:58:13.444 [pimatic-homeduino] debug: received: [ 598, 1875, 3785, 7610 ] 0101010201010201020101020101010202010201010101020201010102010101020102010201010101010303
22:58:13.863 [pimatic-homeduino] debug: data: "RF receive 591 1883 3788 7612 0 0 0 0 0101010201010201020101020101010202010201010101020201010102010101020102010201010101010303"
22:58:13.879 [pimatic-homeduino] debug: received: [ 591, 1883, 3788, 7612 ] 0101010201010201020101020101010202010201010101020201010102010101020102010201010101010303

22:58:18.358 [pimatic-homeduino] debug: data: "RF receive 596 1877 3787 7618 0 0 0 0 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303"
22:58:18.374 [pimatic-homeduino] debug: received: [ 596, 1877, 3787, 7618 ] 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303
22:58:18.811 [pimatic-homeduino] debug: data: "RF receive 583 1891 3800 7637 0 0 0 0 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303"
22:58:18.829 [pimatic-homeduino] debug: received: [ 583, 1891, 3800, 7637 ] 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303

22:58:21.358 [pimatic-homeduino] debug: data: "RF receive 598 1872 3788 7612 0 0 0 0 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303"
22:58:21.375 [pimatic-homeduino] debug: received: [ 598, 1872, 3788, 7612 ] 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303
22:58:21.811 [pimatic-homeduino] debug: data: "RF receive 583 1889 3803 7633 0 0 0 0 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303"
22:58:21.827 [pimatic-homeduino] debug: received: [ 583, 1889, 3803, 7633 ] 0101010201010201020101020101010202010201010101020201010201010101020102010201020201020303

I wanted to switch over to pimatic using my existing sensors. Maybe there is a possibility to implement this sensor?

sysdev41, 2015-03-20:
@Icesory: i think the sensor data cannot be decoded and it is not an issue regarding the configuration file? If that is the case, is the above information sufficient to implement the protocol?

Icesory, 2015-03-20:
Will take a close look at the weekend to this.

sysdev41, 2015-03-20:
That's cool - if you need any more information/debug, just post it an i will try to figure it out ...

xCite1986, 2015-03-23:
since sweetpi changed the pulsecount from 88 to 86 and modify the pulses in weather3.js it got the same error.

i've manuall create a weather31.js with old setup and it works.

https://github.com/pimatic/rfcontroljs/commit/51f8b9317067f78d0b106bcb056415b5972a8117

sweetpi, 2015-03-23:
@xCite1986 Did you try to update homeduino on your arduino?

sysdev41, 2015-03-23:
What should i do now? 
I am using pimatic-homeduino on a raspberry with the recommended bandpass filter: 
https://github.com/pilight/pilight_firmware

xCite1986, 2015-03-23:
pimatic-homeduino was on 0.8.47, pimatic says i'm up to date with my plugins...
```
info [pimatic-cron]: the time is: Mon Mar 23 2015 20:22:49 GMT+0100 (CET)
20:22:48info [pimatic]: Loading plugin: "pimatic-openweather" (0.8.3)
20:22:46info [pimatic]: Loading plugin: "pimatic-mobile-frontend" (0.8.58)
20:22:46info [pimatic]: Loading plugin: "pimatic-shell-execute" (0.8.10)
20:22:46info [pimatic]: Loading plugin: "pimatic-sysinfo" (0.8.12)
20:22:46info [pimatic]: Loading plugin: "pimatic-sunrise" (0.8.0)
20:22:44info [pimatic]: Loading plugin: "pimatic-pushbullet" (0.8.4)
20:22:44info [pimatic]: Loading plugin: "pimatic-ping" (0.8.3)
20:22:44info [pimatic]: Loading plugin: "pimatic-links" (0.8.5)
20:22:43info [pimatic]: Loading plugin: "pimatic-mail" (0.8.3)
20:22:41info [pimatic]: Loading plugin: "pimatic-homeduino" (0.8.47)
20:22:40info [pimatic]: Loading plugin: "pimatic-cron" (0.8.5)
```

```

[pimatic-homeduino]: received: [ 554, 1900, 3794, 7606 ] 0101010101020202010202010201010202020201020201020102020101010101020101010101020101010303
[pimatic-homeduino]: data: "RF receive 554 1900 3794 7606 0 0 0 0 0101010101020202010202010201010202020201020201020102020101010101020101010101020101010303"
[pimatic-homeduino]: received: [ 581, 1908, 3816, 7625 ] 0101020101010101020101010101010102010102010201020201010101020101020201010101020201020303
[pimatic-homeduino]: data: "RF receive 581 1908 3816 7625 0 0 0 0 0101020101010101020101010101010102010102010201020201010101020101020201010101020201020303"
```

Icesory, 2015-03-23:
Sweetpi means the script on the Arduino. Is this up to date?

xCite1986, 2015-03-23:
oh okay :)

this was my weather3.js, i think its the newest...

```
module.exports = function(helper) {
  var protocolInfo, pulsesToBinaryMapping;
  pulsesToBinaryMapping = {
    '01': '0',
    '02': '1',
    '03': ''
  };
  return protocolInfo = {
    name: 'weather3',
    type: 'weather',
    values: {
      id: {
        type: "number"
      },
      channel: {
        type: "number"
      },
      temperature: {
        type: "number"
      },
      humidity: {
        type: "number"
      }
    },
    brands: ["tfa", "conrad"],
    pulseLengths: [508, 2012, 3908, 7726],
    pulseCount: 86,
    decodePulses: function(pulses) {
      var binary, h0, h1, humidity, result, t0, t1, t2, temperature;
      binary = helper.map(pulses, pulsesToBinaryMapping);
      t0 = helper.binaryToNumber(binary, 22, 25);
      t1 = helper.binaryToNumber(binary, 18, 21);
      t2 = helper.binaryToNumber(binary, 14, 17);
      temperature = Math.round(((t0 * 256 + t1 * 16 + t2) * 10 - 12200) / 18) / 10;
      h0 = helper.binaryToNumber(binary, 30, 33);
      h1 = helper.binaryToNumber(binary, 26, 29);
      humidity = h0 * 16 + h1;
      return result = {
        id: helper.binaryToNumber(binary, 2, 9),
        channel: helper.binaryToNumber(binary, 12, 13) + 1,
        temperature: temperature,
        humidity: humidity
      };
    }
  };
};
```
and this my edited weather31.js whitch will work with my sensor

```
module.exports = function(helper) {
  var protocolInfo, pulsesToBinaryMapping;
  pulsesToBinaryMapping = {
    '01': '0',
    '02': '1',
    '03': ''
  };
  return protocolInfo = {
    name: 'weather31',
    type: 'weather',
    values: {
      id: {
        type: "number"
      },
      channel: {
        type: "number"
      },
      temperature: {
        type: "number"
      },
      humidity: {
        type: "number"
      }
    },
    brands: ["tfa", "conrad"],
    pulseLengths: [508, 2012, 3908, 7726],
    pulseCount: 88,
    decodePulses: function(pulses) {
      var binary, h0, h1, humidity, result, t0, t1, t2, temperature;
      binary = helper.map(pulses, pulsesToBinaryMapping);
      t0 = helper.binaryToNumber(binary, 22, 25);
      t1 = helper.binaryToNumber(binary, 18, 21);
      t2 = helper.binaryToNumber(binary, 14, 17);
      temperature = Math.round(((t0 * 256 + t1 * 16 + t2) * 10 - 12200) / 18) / 10;
      h0 = helper.binaryToNumber(binary, 30, 33);
      h1 = helper.binaryToNumber(binary, 26, 29);
      humidity = h0 * 16 + h1;
      return result = {
        id: helper.binaryToNumber(binary, 2, 9),
        channel: helper.binaryToNumber(binary, 12, 13) + 1,
        temperature: temperature,
        humidity: humidity
      };
    }
  };
};
```

result with weather31.js.

```
[pimatic-homeduino]: weather31: { id: 130, channel: 1, temperature: 22.1, humidity: 49 }
[pimatic-homeduino]: received: [ 582, 1909, 3811, 7635 ] 0101020101010101020101010101010102010102010201020201010101020101020201010101020201020303
[pimatic-homeduino]: data: "RF receive 582 1909 3811 7635 0 0 0 0 0101020101010101020101010101010102010102010201020201010101020101020201010101020201020303"
[pimatic-homeduino]: weather31: { id: 29, channel: 3, temperature: 13.4, humidity: 45 }
[pimatic-homeduino]: received: [ 556, 1900, 3791, 7614 ] 0101010101020202010201020201010202010201020201020102020201020101020101010101020201020303
[pimatic-homeduino]: data: "RF receive 556 1900 3791 7614 0 0 0 0 0101010101020202010201020201010202010201020201020102020201020101020101010101020201020303"

```

Icesory, 2015-03-23:
@sweetpi this is really strange. Why are the Arduino and the GPIO receivers are different?
Have you forgot to copy the new RFControl to the vhduino?
This is also affected by this bug
http://forum.pimatic.org/topic/59/auriol-temperature-sensor-type-ian-91838-not-detected/3

sweetpi, 2015-03-23:
@Icesory You're right, vhduino uses the old buggy version (https://github.com/pimatic/virtualhomeduino). I will update the binary.

sysdev41, 2015-03-24:
Do i still have to use the modified weather31.js file @xCite1986 has posted, or is it sufficient to wait for the updated RFControl binary/library in virtualhomeduino?

xCite1986, 2015-03-24:
you could still modify the file with nano byself, but if your pimatic-homeduino gets an update, you will fall back to original weather3.js

edit with ...

sudo nano pimatic-app/node_modules/pimatic-homeduino/node_modules/homeduino/node_modules/rfcontroljs/lib/protocols/weather3.js

```
pulseCount: 86
```
to 

```
pulseCount: 88
```

... would work temporary

Icesory, 2015-03-24:
With the next update this problem should be fixed. 
We have changed the verification system in RFControl massive. But this update isn't in the gpio driver at the moment. Every user how used the gpio instead a Arduino used the old version. 
The problem is that we have a weather sensor that is only receivable by the old gpio system. Now we must look why and then we can create a bug fix.

sysdev41, 2015-03-24:
Thanks a lot, it is working with the "quick fix" :-) ! 
@Icesory do you know when this update will take place (i mean days/weeks/months)? Do i have to apply the "quick fix" each time, i get pimatic updates, or just when updates arrive for pimatic-homeduino plugin?

Icesory, 2015-03-24:
You have to fix this only by an update of pimatic-homeduino.
I think this should be fixed in the next two weeks.

Icesory, 2015-04-01:
How is the actual state here? vhduino(the GPIO driver) was fixed with the last homeduino update.
Worked it now as expected?

xCite1986, 2015-04-01:
hey @icestory.

yes, weather3 works now for me. Thanks for quickly fixing

closed 2015-04-01 by sweetpi

sysdev41, 2015-04-02:
Hi Icesory,
thanks for the homeduino update. Just tested it, and it now works out of the box! Great!!

Icesory, 2015-04-02:
All credits goes to @sweetpi. He is the master mind behind this.