pimatic/rfcontroljs #46: Help porting protocol from pilight

kirichkov, 2015-03-19:
I've written a protocol for pilight for the RSL366 switches (apparently also sold as PROmax) in pilight - [link to pilight wiki](http://wiki.pilight.org/doku.php/rsl366).

I'm trying to port the protocol to pimatic, however, I'm unable to get it to work. 

Some background on my setup:
I use the cheap RF sender (no receiver as I don't have my switches' remote) connected directly to the Pi's GPIO pin. (WiringPi: 0)
I am able to use pilight v6.0 to switch the devices on and off without any issue.
I used the "arctech_switch_old" protocol with pilight before my protocol was added to the master and tried to simply port the unit/id configuration - that didn't work and after some googling I found that these are not always the same in pimatic - https://github.com/pimatic/pimatic/issues/380.

I've gotten to the point to give the commands directly to the vhduino, but I'm having trouble figuring out what the "[buckets](https://github.com/pimatic/virtualhomeduino/blob/master/vhduino.cpp#L157)" are.

I've gotten to the point to construct the following string and feed it to vhduino:
    RF send 0 5 390 1170 13260 0 0 0 0 0 01100110011001010110011001010110011001100110010102

This should be switching unit 4-3 on. 

I'm also uncertain on what the five zeroes between 13260 and the beginning of the pulses are supposed to do and whether these are correct values for the switches.

Icesory, 2015-03-20:
When you read the readme from rfcontroljs then it should clear all questions for you. 
https://github.com/pimatic/rfcontroljs

kirichkov, 2015-03-20:
I definitely skipped that first part! Thanks!

In the meantime I figured out why my switches were not working - I missed the "translation table", so I think it's better to create another translation table for the device, instead of adding yet another protocol that resembles a lot the one in switch4. After all the Unit/IDs in pimatic have different meaning than the dials.

closed 2015-03-23 by kirichkov

Sennevds, 2016-10-03:
Hi @kirichkov,

do you still have protocol?