pimatic/rfcontroljs #68: Add support for the Hideki/UPM protocol for Hama TS33C and other sensors

mwittig, 2015-08-30:
Unfortunately, the signal sent by sensor is not picked up by hemduino (no debug log output). According to http://forum.pilight.org/Thread-hama-ts33c-temperature-hygrometer the TS33C sensor is using the Hideki/UPM protocol used by various OEM products such as TFA Nexus, Mebus, Irox, Irox-Pro X, Honeywell, Cresta TE923, TE923W, TE821W,
WXR810, DV928, Ventus W906. Another thread indicates the protocl has been reverse engineered, see http://www.wetterstationen.info/forum/entwicklerforum/tx-protokoll-offen-te923-cresta815-irox-(hideki)

The protocol document can be found here http://members.upc.nl/m.beukelaar/Crestaprotocol.pdf

See also http://forum.pimatic.org/topic/807/hama-wheater-sensor-not-showing-up-in-debug-logs

Icesory, 2015-09-15:
For this we need an other verification system in the arduino.
I had startet the development of a simpler version of the actual system. Eventually i am able to implement this. But this needs also a much additions to pimatic-homeduino.

mwittig, 2015-09-16:
Thanks for the reply! Haven't seen you online for a couple weeks! We have been missing you - on homeduino affairs, in particular. Several users are struggling with homeduino and currently none of the experts is around. 

It is great you have already started working on an improved verification system and it is pity you are currently busy with other things. If I can help you at some point please let me know.  I have no knowledge of the verification system so far but may be I can help in other areas.