pimatic/rfcontroljs #80: Error with Led2 device

renearts, 2016-02-09:
Since (I think) yesterday I get an error on a Led2 device which has until now worked fine. I configured it half a year ago and haven't had any real trouble with it. 
Error loading device rfswitch-ledstrip-keuken: "led2" must be one of the following types: "switch, dimmer" 
Is the Led2 protocol not recognised as dimmer anymore?

The device configuration:
`{
      "id": "rfswitch-ledstrip-keuken",
      "name": "LEDstrip keuken",
      "class": "HomeduinoRFDimmer",
      "protocols": [
        {
          "name": "led2",
          "options": {
            "id": 2977,
            "command": "on/off"
          },
          "send": true,
          "receive": false
        }
      ]
    },`

Icesory, 2016-02-09:
We have published yesterday a stricter config check. So every wrong config for Homeduino will now be shown. The led2 protocol has no support for dimmers. You can use it only with an Buttons device.

renearts, 2016-02-09:
Ok, that sounds reasonable, though the device I use actually is a really cheapo Chinese PITA inline dimmer (dimming + several effects do work with the remote control). I don't use the dimming or the effects functionality though, so I would be perfectly fine with only on/off functionality for now. I want to replace the dimmer sometime soon anyway. 
I'd like to get stuff working again for the time being; how should I adapt my configuration?

When I alter the class to HomeDuinoRFSwitch it throws: Error loading device rfswitch-ledstrip-keuken: "led2" must be one of the following types: "switch"

Icesory, 2016-02-09:
You must use the class HomeduinoRFButtonsDevice. Please look at the Homeduino readme to see exactly how it must be configured.

renearts, 2016-02-09:
Thanks, wasn't able to find the correct readme that fast. But it all works now, so thanks a lot!

closed 2016-02-09 by renearts

Icesory, 2016-02-10:
No problem. You're welcome.