pimatic/rfcontroljs #81: After update pimatic-homeduino (0.8.67)  to  pimatic-homeduino (0.8.69)

WimUtTwente, 2016-02-09:
Error loading device Ben_st_lamp: Protocol "rolling1" has no property named "codeOn". Available properties are: "code"

    {
      "id": "Ben_st_lamp",
      "name": "Beneden_St_lamp",
      "class": "HomeduinoRFSwitch",
      "protocols": [
        {
          "name": "rolling1",
          "options": {
            "codeOn": [
              "001100000110011110101100",
              "001100101001000000011100",
              "001110110001101101101100",
              "001111101010001101011100"
            ],
            "codeOff": [
              "001100110010110101111100",
              "001101111101010010111100",
              "001110010100100000101100",
              "001111001000000110001100"
            ]
          }
        }
      ]
    },

Icesory, 2016-02-09:
Thanks for the report. This is really a bug. Will look how I can fix this. Sorry for the trouble.

closed 2016-02-10 by Icesory

WimUtTwente, 2016-02-10:
Everything works again thanks for the quick response.