pimatic/rfcontroljs #87: RGB LED strip

mauricesundermann, 2016-03-16:
Hi folks,

Just started with Pimatic and the rfcontroljs on my Raspberry Pi. Normal (KAKU) rf switches work but my RGB strip give issues.
As i saw in this post [https://github.com/pimatic/rfcontroljs/issues/42](url), i added my RGB LED strip from Ebay. 

```
"id": "ledstriptuin",
      "name": "Tuin_RGBLED_olijf",
      "class": "HomeduinoRFButtonsDevice",
      "buttons": [
        {
          "id": "on-button",
          "text": "Aan",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "code:00010011",
                "id": 14754
              }
            }
          ]
        },
        {
          "id": "off-button",
          "text": "Uit",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "code:00010011",
                "id": 14754
              }
            }
          ]
        },
        {
          "id": "buttonRed",
          "text": "Rood",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "code:00000010",
                "id": 14754
              }
            }
          ]
        },
        {
          "id": "buttonGreen",
          "text": "Groen",
          "protocols": [
            {
              "name": "led4",
              "options": {
                "command": "code:00001000",
                "id": 14754
              }
            }
          ]
        }
      ]
    }
  ],
```

I got the following error in pimatic:
```
error [pimatic]: Error loading device "ledstriptuin": Protocol "led4" can´t handle the command "code:00010011". Available commands are: "on/off, bright+, bright-, color-, color+"
error [pimatic]: Error loading device "ledstriptuin": Protocol "led1" can´t handle the command "code:00010011". Available commands are: "on/off, up, down"
```

As you can see i tried with led 4 and led 1 protocol. I coppied the command codes from the log messages window. 
What am i doing wrong? Thank you in advance

Icesory, 2016-03-16:
You config is generally right. I have implemented a check and this don't handle the "code:xxx" option. 
When you press a button on your remote, did you see some debug messages. Normally the "code:" pattern isn't needed.

mauricesundermann, 2016-03-17:
This are the debug logs.
Power toggle switch:
`debug [pimatic-homeduino]: doorbell1:  { id: 2659, unit: 891, state: true }
10:19:06debug [pimatic-homeduino]: led4:  { id: 14754, command: 'code:00010011' }
10:19:06debug [pimatic-homeduino]: led3:  { id: 14754, command: 'yellow' }
10:19:06debug [pimatic-homeduino]: led2:  { id: 14754, command: 'code:00010011' }
10:19:06debug [pimatic-homeduino]: led1:  { id: 14754, command: 'code:00010011' }
10:19:06debug [pimatic-homeduino]: contact3:  { id: 812510, contact: false }
10:19:06debug [pimatic-homeduino]: contact2:  { id: 812510, contact: false }
10:19:06debug [pimatic-homeduino]: switch22:  { id: 812510, command: undefined }
10:19:06debug [pimatic-homeduino]: switch15:  { id: 236065, unit: 2, state: false, all: false }
10:19:06debug [pimatic-homeduino]: switch14:  { id: 236065, unit: 4, all: false, state: true }
10:19:06debug [pimatic-homeduino]: switch5:  { id: 812510, unit: 0, all: true, state: false }
10:19:06debug [pimatic-homeduino]: received: [ 316, 932, 9384 ] 01011010100101101001100101011001010101100101101002
10:19:06debug [pimatic-homeduino]: data: "RF receive 316 932 9384 0 0 0 0 0 01011010100101101001100101011001010101100101101002"`

Red:
`debug [pimatic-homeduino]: doorbell1:  { id: 2659, unit: 1019, state: false }
10:20:20debug [pimatic-homeduino]: led4:  { id: 14754, command: 'code:00000010' }
10:20:20debug [pimatic-homeduino]: led3:  { id: 14754, command: 'code:00000010' }
10:20:20debug [pimatic-homeduino]: led2:  { id: 14754, command: 'code:00000010' }
10:20:20debug [pimatic-homeduino]: led1:  { id: 14754, command: 'code:00000010' }
10:20:20debug [pimatic-homeduino]: contact3:  { id: 812511, contact: true }
10:20:20debug [pimatic-homeduino]: contact2:  { id: 812511, contact: false }
10:20:20debug [pimatic-homeduino]: switch22:  { id: 812511, command: 'B' }
10:20:20debug [pimatic-homeduino]: switch15:  { id: 236064, unit: 0, state: true, all: true }
10:20:20debug [pimatic-homeduino]: switch14:  { id: 236064, unit: 4, all: false, state: false }
10:20:20debug [pimatic-homeduino]: switch5:  { id: 812511, unit: 0, all: true, state: true }
10:20:20debug [pimatic-homeduino]: received: [ 316, 936, 9380 ] 01011010100101101001100101011001010101010101100102
10:20:20`

Green:
`debug [pimatic-homeduino]: doorbell1:  { id: 2659, unit: 1787, state: false }
10:20:54debug [pimatic-homeduino]: led4:  { id: 14754, command: 'code:00001000' }
10:20:54debug [pimatic-homeduino]: led3:  { id: 14754, command: 'demo' }
10:20:54debug [pimatic-homeduino]: led2:  { id: 14754, command: '50%' }
10:20:54debug [pimatic-homeduino]: led1:  { id: 14754, command: 'code:00001000' }
10:20:54debug [pimatic-homeduino]: contact3:  { id: 812511, contact: true }
10:20:54debug [pimatic-homeduino]: contact2:  { id: 812511, contact: false }
10:20:54debug [pimatic-homeduino]: switch22:  { id: 812511, command: 'D' }
10:20:54debug [pimatic-homeduino]: switch15:  { id: 236064, unit: 0, state: false, all: true }
10:20:54debug [pimatic-homeduino]: switch14:  { id: 236064, unit: 3, all: true, state: false }
10:20:54debug [pimatic-homeduino]: switch5:  { id: 812511, unit: 0, all: true, state: true }
10:20:54debug [pimatic-homeduino]: received: [ 316, 936, 9380 ] 01011010100101101001100101011001010101011001010102
10:20:54debug [pimatic-homeduino]: data: "RF receive 316 936 9380 0 0 0 0 0 01011010100101101001100101011001010101011001010102"`

Blue:
`debug [pimatic-homeduino]: doorbell1:  { id: 2659, unit: 251, state: false }
10:21:21debug [pimatic-homeduino]: led4:  { id: 14754, command: 'code:00001110' }
10:21:21debug [pimatic-homeduino]: led3:  { id: 14754, command: 'white' }
10:21:21debug [pimatic-homeduino]: led2:  { id: 14754, command: 'code:00001110' }
10:21:21debug [pimatic-homeduino]: led1:  { id: 14754, command: 'code:00001110' }
10:21:21debug [pimatic-homeduino]: contact3:  { id: 812511, contact: true }
10:21:21debug [pimatic-homeduino]: contact2:  { id: 812511, contact: false }
10:21:21debug [pimatic-homeduino]: switch22:  { id: 812511, command: undefined }
10:21:21debug [pimatic-homeduino]: switch15:  { id: 236064, unit: 4, state: true, all: false }
10:21:21debug [pimatic-homeduino]: switch14:  { id: 236064, unit: 1, all: false, state: false }
10:21:21debug [pimatic-homeduino]: switch5:  { id: 812511, unit: 1, all: false, state: false }
10:21:21debug [pimatic-homeduino]: received: [ 316, 940, 9388 ] 01011010100101101001100101011001010101011010100102
10:21:21debug [pimatic-homeduino]: data: "RF receive 316 940 9388 0 0 0 0 0 01011010100101101001100101011001010101011010100102"`

White:
`debug [pimatic-homeduino]: doorbell1:  { id: 2659, unit: 1403, state: false }
10:22:06debug [pimatic-homeduino]: led4:  { id: 14754, command: 'code:00010100' }
10:22:06debug [pimatic-homeduino]: led3:  { id: 14754, command: 'cyan' }
10:22:06debug [pimatic-homeduino]: led2:  { id: 14754, command: 'code:00010100' }
10:22:06debug [pimatic-homeduino]: led1:  { id: 14754, command: 'code:00010100' }
10:22:06debug [pimatic-homeduino]: contact3:  { id: 812510, contact: true }
10:22:06debug [pimatic-homeduino]: contact2:  { id: 812510, contact: false }
10:22:06debug [pimatic-homeduino]: switch22:  { id: 812510, command: 'C' }
10:22:06debug [pimatic-homeduino]: switch15:  { id: 236065, unit: 0, state: true, all: true }
10:22:06debug [pimatic-homeduino]: switch14:  { id: 236065, unit: 2, all: true, state: false }
10:22:06debug [pimatic-homeduino]: switch5:  { id: 812510, unit: 0, all: true, state: true }
[... cut ...]  [pimatic-homeduino]: received: [ 320, 936, 9368 ] 01011010100101101001100101011001010101100110010102
10:22:06debug [pimatic-homeduino]: data: "RF receive 320 936 9368 0 0 0 0 0 01011010100101101001100101011001010101100110010102"`

And there are more buttons, see the ebay item here: [http://www.ebay.com/itm/20-Key-RF-Wirless-Remote-LED-Controller-12-24V-For-5050-3528-RGB-LED-Strip-Light-/290912753414?hash=item43bbc05706](url)
Please let me know if you need the debug codes from the other buttons.

Icesory, 2016-03-17:
You can use all buttons, which are decoded with an led protocol and a unique command. 
For on/off you can use led3 and "yellow"

mauricesundermann, 2016-03-17:
But what can I use for red? Or other buttons that arn't decoded?
Isn't it possible to make a led 5? If you let me, I will give you all the codes of the buttons. It's quite a populair device on ebay. I can't imagine i'm the only one who want to use it with Pimatic. It can't be that hard becouse if you look to LED4, the code is pritty simple and always the same when pressed.

Icesory, 2016-03-18:
Clearly you can create a led5 protocol. 
Or you post all buttons with the corresponding rf receive messages.

mauricesundermann, 2016-03-22:
Thank you for your reply. 
This are the RF receive messages:
`
Power toggle:
"RF receive 324 936 9412 0 0 0 0 0 01011010100101101001100101011001010101100101101002"

Play / pause:
"RF receive 316 936 9404 0 0 0 0 0 01011010100101101001100101011001010101011010011002"

Brightness up:
"RF receive 316 932 9408 0 0 0 0 0 01011010100101101001100101011001010101010101011002"

Brightness down:
"RF receive 316 940 9424 0 0 0 0 0 01011010100101101001100101011001010101010110101002"

Red:
"RF receive 320 940 9436 0 0 0 0 0 01011010100101101001100101011001010101010101100102"

Green:
"RF receive 320 940 9432 0 0 0 0 0 01011010100101101001100101011001010101011001010102"

Blue:
"RF receive 320 944 9452 0 0 0 0 0 01011010100101101001100101011001010101011010100102"

White:
"RF receive 320 940 9444 0 0 0 0 0 01011010100101101001100101011001010101100110010102"

Orange:
"RF receive 320 940 9440 0 0 0 0 0 01011010100101101001100101011001010101010101101002"

Yellow:
"RF receive 320 944 9448 0 0 0 0 0 01011010100101101001100101011001010101011001011002"

Cyan:
"RF receive 320 948 9448 0 0 0 0 0 01011010100101101001100101011001010101011010101002"

Purple:
"RF receive 320 944 9452 0 0 0 0 0 01011010100101101001100101011001010101100110011002"

Speed plus:
"RF receive 320 944 9428 0 0 0 0 0 01011010100101101001100101011001010101100110100102"

Speed min:
"RF receive 320 944 9452 0 0 0 0 0 01011010100101101001100101011001010101100110101002"

Auto:
"RF receive 320 936 9440 0 0 0 0 0 01011010100101101001100101011001010101010110010102"

Flash:
"RF receive 320 944 9440 0 0 0 0 0 01011010100101101001100101011001010101010110011002"

Jump 3:
"RF receive 320 940 9440 0 0 0 0 0 01011010100101101001100101011001010101011001100102"

Jump 7:
"RF receive 320 944 9436 0 0 0 0 0 01011010100101101001100101011001010101011001101002"

Fade 3:
"RF receive 320 944 9456 0 0 0 0 0 01011010100101101001100101011001010101100101010102"

Fade 7:
"RF receive 320 944 9440 0 0 0 0 0 01011010100101101001100101011001010101100101011002"
`

Please let me know if you need more info :)