pimatic/rfcontroljs #96: Test script not working?

rrooggiieerr, 2016-05-12:
Hi,

I'm trying to implement a new protocol for my projector screen and think I have the testenvironment set up (lots of dependencies!)

Before I start developing I run the test script to see if it works and am getting all kinds of errors.
```
gulp test
[08:35:17] Using gulpfile ~/rfcontroljs/gulpfile.js
[08:35:17] Starting 'test'...

  #decodePulses()
    ✓ generic should decode the pulses
    ✓ generic2 should decode the pulses
    ✓ pir1 should decode the pulses
Warning trying to parse message with protocol switch2: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pimatic/rfcontroljs/src/protocols/switch2.coffee:27:23)
Warning trying to parse message with protocol switch3: Data did not match mapping
  at Object.protocolInfo.decodePulses (/home/pimatic/rfcontroljs/src/protocols/switch3.coffee:27:23)
...
...
  77 passing (7s)
  2 failing

  1) #decodePulses() pir2 should decode the pulses:
     Error: timeout of 2000ms exceeded. Ensure the done() callback is being called in this test.
  

  2) #decodePulses() doorbell3 should decode the pulses:

      AssertionError: pulse of doorbell3 should be detected.
      + expected - actual

      -false
      +true
      
    at Context.runTest (test/lib-controller.coffee:750:9)
```

Is the test script actually maintained, or is it deprecated?

rrooggiieerr, 2016-05-12:
Ok, seems like those warnings can be ignored, but the #decodePulses() pir2 and #decodePulses() doorbell3 tests fail.

sweetpi, 2016-08-21:
Yes the warnings can be ignored. I've fixed doorbell3.

closed 2016-08-21 by sweetpi