pimatic/rfcontroljs #101: Add support for the wp515s doorbell protocol

primatic, 2016-05-25:
I reconstructed the protocol using a 433 receiver hooked up to the mic input of my computer to record the pulse sequences. From that I constructed the protocol.

Using the homeduino transmission works, but reception fails.

Using the simulate tool of the homeduino and protocol timing as
295, 590,295,590,295,590,295,590,295, 295,590,295,590,295,590,590,295,295,590,590,295,295,590,9864,
295, 590,295,590,295,590,295,590,295, 295,590,295,590,295,590,590,295,295,590,590,295,295,590,9864,
...
reveals that it restarts detecting at
rec_pos= 23 pack_pos= 23 duration= 2466 => restart package => start recoding

no clue why....
I did not do a deep dive into the homeduino code, but others seem to have problems with the current implementation of the doorbell3 protocol (for which a pull is still pending)

primatic, 2016-06-03:
Well, since it somehow bothers me that I did not get it to work, I spent some time digging into it....
I changed the number of packets in RFControl.to be **16** or more (was 32), and with the protocol timings as listed above, _simulate_ now says

Package are equal.
result:
292 588 292 588 292 588 292 588 292 292 588 292 588 292 588 588
292 292 588 588 292 292 588 9864
compressed: 292 588 9864 0 0 0 0 0  t: 010101010010101100110012

seems that I found a piece of the puzzle.

primatic, 2016-06-03:
in pimatic, enabled debug messages, hit the (physical) doorbell button and got a lot of these messages:
`debug [pimatic-homeduino]: data: "RF receive 300 580 10228 0 0 0 0 0 01010101001010101100110012"
debug [pimatic-homeduino]: received: [ 300, 580, 10220 ] 01010101001010101100110012`

looks like reception of the protocol no succeeds

primatic, 2016-06-05:
fix for doorbell3 test is part of pull request

mwittig, 2016-07-28:
I think this one has been solved now. Please re-open if it is not the case or you whish to follow-up on this. Thanks

closed 2016-07-28 by mwittig