pimatic/rfcontroljs #105: Require assistance

PJCzx, 2016-07-27:
Hi guys,

Sorry to bother but I'm quite a noob in most of the subjects here and I'm a bit lost. I'm trying to replicate the signal of the remote control for projection screen.

I tried many lib, but only RFControl look to give me feedback.

```
b: 612 244 144 5204 0 0 0 0 
t: 001200120202001012020012020010101012001010101200120012020202001012020202020201001202010202020202020200120200120202001201001012010013
```

I understand the **pulse lengths** are [144, 244, 612, 5204] (not exactly doubled each times :s). The **pulse count** is 132 and the **pulse sequence** is written above.

The protocols with a pulse count of 132 are:
- contact1.js [268, 1282, 2632, 10168]
- pir5 [268, 1282, 2632, 10168]
- switch1.js [268, 1282, 2632, 10168]
- switch17.js [260, 2680, 1275, 10550]

None of those have a close pulse length.

I'd like hints to:
1. Identify the next steps
2. Get a few instruction on which file to edit, where and how to run tests with gulp (not very familliar)

I opened a [Stackoverflow](http://stackoverflow.com/questions/37667102/clone-433-92mhz-remote-raspberry-pi-arduino) without answer since weeks, I'm now trying to contact any of you.

Thank you in advance, once again sorry to rise an issue for that.

Best,
PJ

mwittig, 2016-07-28:
Hi, in which context are you planning to use RFControlJS? Are you using pimatic? If yes it might be helpful to trace debug messages which can be produced by pimatic-homeduino on request by setting the "debug" property as part of the plugin configuration.

PJCzx, 2016-07-28:
Hello, thx for your answer. Because I'm used to it, I was more about to create an API or an [Hombridge ](https://github.com/nfarina/homebridge) plugin. What could be the easiest way ? I need to "learn" the remote (2 buttons) from the Rx, and then being able to send back the signals with the Tx.

PJCzx, 2016-07-29:
Thx to your feedback I'm following this [tuto](https://forum.pimatic.org/topic/210/5-homeduino-and-rf-devices-433-mhz-transmitting-receiving). I hope to make progress. I will keep you in touch. Thx again

PJCzx, 2016-08-01:
Hello @mwittig,

I've been through the tutorial as far as you mentioned; I'm now getting:

```info [pimatic-homeduino]: Connected to homeduino device.```

Nothing is displays when I press the remote buttons. I'm really close to the RPI, the projection screen is going up and down and I display all logs levels for the pimatic-homeduino plugin. Others logs give success information.

Should something happen even if the protocol is not recognized ? Do you have a hint?

Thx

PJCzx, 2016-08-09:
Idem, I don't know how to define `binaryToPulse`. Does anyone have good documentation on this?