pimatic/rfcontroljs #112: Add weather 19 Landmann BBQ Thermometer

Flop2006, 2016-09-12:
See here: https://forum.pimatic.org/topic/1768/protocol-for-thermometer
weather19.js:
```
module.exports = function(helper) {
  var protocolInfo, pulsesToBinaryMapping;
  pulsesToBinaryMapping = {
    '01': '0',
    '02': '1',
    '03': ''
  };
  return protocolInfo = {
    name: 'weather19',
    type: 'weather',
    values: {
      temperature: {
        type: "number"
      },
      id: {
        type: "number"
      },
      channel: {
        type: "number"
      },
    },
    brands: ["Landmann BBQ Thermometer"],
    pulseLengths: [548, 1008, 1996, 3936],
    pulseCount: 66,
    decodePulses: function(pulses) {
     /*
      Pulse like:
      |020202010101|0101|01010101010101020101010102010201|0201010102010202|03|
      | 1 1 1 0 0 0| 0 0| 0 0 0 0 0 0 0 1 0 0 0 0 1 0 1 0| 1 0 0 0 1 0 1 1|  |
      |ID          | CH |            Temperature         |?               |  |
      |56          |  1 |            266(26.6)           |?               |  |
    */
      var binary, result;
      binary = helper.map(pulses, pulsesToBinaryMapping);
      return result = {
        id: helper.binaryToNumber(binary, 0, 5),
        temperature: helper.binaryToSignedNumber(binary, 8, 23) / 10,
        channel: helper.binaryToNumber(binary, 6, 7) + 1,
      };
    }
  };
};
```

weather19.coffee:
```
module.exports = (helper) ->
  pulsesToBinaryMapping = {
    '01': '0' #binary 0
    '02': '1' #binary 1
    '03': ''  #footer
  }
  return protocolInfo = {
    name: 'weather19'
    type: 'weather'
    values:
      temperature:
        type: "number"
      id:
        type: "number"
      channel:
        type: "number"
    brands: ["Landmann BBQ Thermometer"]
    pulseLengths: [548, 1008, 1996, 3936]
    pulseCount: 66
    decodePulses: (pulses) ->
      # pulses is something like: '020202010101010101010101010101020101010102010201020101010201020203'
[... 231 lines cut ...]
      throw new Error("Could not find a protocol named " + protocolName);
    }
    if (protocol.encodeMessage == null) {
      throw new Error("The protocol has no send report.");
    }
    return {
      pulses: protocol.encodeMessage(message),
      pulseLengths: protocol.pulseLengths
    };
  },
  getAllProtocols: function() {
    return protocols;
  },
  getProtocol: function(protocolName) {
    var p, _i, _len;
    for (_i = 0, _len = protocols.length; _i < _len; _i++) {
      p = protocols[_i];
      if (p.name === protocolName) {
        return p;
      }
    }
    return null;
  }
};
```

mwittig, 2016-09-12:
Excellent! Thank you very much for your contribution. I am going to integrate this asap.

mwittig, 2016-09-14:
released rfcontroljs@0.0.56, homeduino@0.0.62, pimatic-homeduino@0.9.10

closed 2016-09-14 by mwittig