pimatic/rfcontroljs #123: Switch3 (Brennstuhl) switching wrong way

FROeHlyEisvogel, 2017-11-28:
Hey there,
I am using Pimatic with the Homeduino plugin for swiching my remote sockets.
The pruducer is "Brennstuhl" so I use the switch3 protocol which works fine.

There is just one small issue, the switching works the wrong way.
Switching on in the Pimatic GUI turns off the socket and switching off in the GUI turns the socket on.

mwittig, 2017-11-28:
That's interesting as Brennstuhl has been reported and documented to properly work with `switch3`. Please use `switch2` instead. That should fix it. If not, please re-open this ticket.

Please also send a comment with exact model number of the socket you have.

closed 2017-11-28 by mwittig

FROeHlyEisvogel, 2017-11-29:
Thank you for your help.
Switch2 works fine.
The socket I am using is "RCS 1000 N".